mavlink/qgroundcontrol #1167: Crash when opening serial port

Susurrus, 2015-01-09:
If I open QGC, then select COM4 and 115200 baud in the upper left and hit the connect button, sometimes QGC crashes. If I can get the bug to trigger in the debugger, I'll try and follow up with it.

Windows 8 x64 (32-bit build on Qt5.4.1.

Susurrus, 2015-01-09:
And then if I restart QGC and try to connect again, I get an "Access is denied" error prompting a restart before I can actually open the port.

Susurrus, 2015-03-10:
Haven't seen this since the toolbar/connection rework and my commits from #1342.

closed 2015-03-10 by Susurrus

kirankotari, 2016-09-13:
I am facing same "Access is denied" for windows 7 and Ubuntu. Is there any solution ?

DonLakeFlyer, 2016-09-13:
Have you installed the drivers on Windows and do you have correct permissions set up on Ubuntu?