mavlink/qgroundcontrol #1168: Live data plot doesn't move viewport when new data fields are selected

Susurrus, 2015-01-09:
I was tracking the `SYSTEM_TIME.time_usec` field in the live data display. Then I switched over to the `LOCAL_POSITION_NED.x` field, with a value of something like 65. The `time_usec` field was up at 1.4e9, and there's no way to shift the viewport down to see the now-small-valued data. To see this data now requires a restart of QGC.

Tagged as important since restarting QGC is the only workaround.

thomasgubler, 2015-01-09:
I played around with this. For me the issue only appears if you zoom in and then switch to another field. It does not matter if you zoom out again (using a right click) before switching. The view stays at the same location.

Switching without zooming seems to work fine.

Susurrus, 2015-01-09:
Would improved behavior be to zoom all the way out if the field selection changes or to just alter the "zoomed out" bounds when the fields are changed to make sure it includes all of the plotted data?

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier