mavlink/qgroundcontrol #1173: Logging files should have .mavlink appended

Susurrus, 2015-01-14:
When saving a datalog from QGC, you need to manually specify the .mavlink, even though QGC shows that this is the filetype you will be saving. QGC should detect if the user has added the .mavlink ending and automatically add it if they haven't.

thomasgubler, 2015-01-14:
This sounds like something the file save dialog should handle. Whoever
picks this up should check if the work is not already done for us and just
needs activation.

Susurrus, 2015-01-14:
Looks like the relevant code section is QGCApplication.cc:479, which doesn't have any file ending checking. It'll just need a few extra lines of code.

dogmaphobic, 2015-01-28:
Qt calls it "suffix". This can be either done by creating a QFileDialog object (instead of using the static method) and setting the default suffix, or processing the returned filename using something like QFileInfo. I would like, if it's OK, to take this as a first exercise in code submission. This is the first time I'm using github and I'm trying to wrap my head around its workings.

Susurrus, 2015-01-28:
Please go ahead, and feel free to assign it to yourself on the right here.

dogmaphobic, 2015-01-28:
Will do. How do I get added as a collaborator? I can't seem to be able to assign myself to the issue unless I'm authorized as a collaborator. Sorry if this is basic but as I said, I'm learning github as I go. Thanks!

Susurrus, 2015-01-28:
I think you get in that list once you've had a PR accepted or something like that. Just go ahead and work on this, nobody else will take it.

dogmaphobic, 2015-01-28:
Got it. I have the fix done. It took me about 10 minutes to add the default suffix, update the upstream classes and reflect the changes into unit test. It has taken me 3 hours figuring out how to get these to my own fork though :) I'm just afraid to mess up the master repo. No worries, I will figure this out.

Thanks!

Susurrus, 2015-01-28:
You should always work on a branch of your repo (`checkout -b NEW_BRANCH_NAME`) when working so that it's easy to reorder commits and put them on a new branch. And then submit a pull request with that branch for review, even if it's a work in progress. Makes it easier for us to comment on it. If those commits end up being bad, you can just fix them up on a new branch and submit a new PR.

thomasgubler, 2015-01-30:
This issue actually applies to all file save dialogs. For example the parameter save dialog has the same problem.

dogmaphobic, 2015-01-30:
> This issue actually applies to all file save dialogs. For example the parameter save dialog has the same problem.

Yes, indeed I've changed them all. Technically it's all done and ready. I'm still sorting out how to unit test these and also, how to properly use doxygen markup.

Susurrus, 2015-01-31:
Fixed.

closed 2015-01-31 by Susurrus