mavlink/qgroundcontrol #1177: Compilation issue on Master branch

jayceelock, 2015-01-20:
Hi, I'm having trouble compiling the new version of QGroundcontrol. The make command keeps producing the error

    In file included from src/ui/designer/QGCToolWidget.cc:12:0:
    src/ui/designer/QGCParamSlider.h:6:44: fatal error: QtDesigner/QDesignerExportWidget: No such file or directory
     #include <QtDesigner/QDesignerExportWidget>
                                            ^
    compilation terminated.
    make: *** [build-release/obj/QGCToolWidget.o] Error 1

The I'm running Mint 17.1 (based on Ubuntu 14.10) and my qmake --version commands outputs
    QMake version 3.0 
    Using Qt version 5.3.2 in /opt/qt53/lib

qmake also produces no errors when its run. Am I doing something wrong?

thomasgubler, 2015-01-20:
How did you install qt 5.3?

If you were using https://launchpad.net/~beineri/+archive/ubuntu/opt-qt532-trusty did you follow the instructions on the top of that page (about sourcing the script) ?

jayceelock, 2015-01-20:
Yes I did. As you can see from the qmake --version argument, it points to /opt/qt53/lib which is the default location where the ppa installs qt5.3. Should I maybe try qt5.4 instead?

thomasgubler, 2015-01-20:
5.3 should work

* how do you run make? From a terminal or qtcreator? --> are you sure that the environment variables are set when running make?
* does the file actually exist? Please check with `find /opt/qt53 -name QDesignerExportWidget`

jayceelock, 2015-01-21:
Hmmmm that seems to have fixed the problem. I couldn't find the `QDesignerExportWidget` file, so I installed the `qt53tools` package from the ppa and now it seems to be compiling fine.

I was informed of the other packages I needed to install (webkits, svg, serialport, etc.) during the `qmake` step. Perhaps the `qt53tools` module should be added to this list of required modules?

thomasgubler, 2015-01-21:
The readme actually mentions qttools5-dev.

Re-reading your original post: since your mint version is based on ubuntu 14.10 why did you need to install qt 5.3 from a ppa? It should be available:  http://packages.ubuntu.com/search?suite=utopic&arch=any&searchon=names&keywords=qt5-default

jayceelock, 2015-01-21:
Ah that's my mistake, I just copied that section without reading through it properly. I'm not too sure. I just updated aptitude again and saw that the available qt5 version is 5.2.1. I'm guessing that Mint might be a bit slow with keeping their repo's synced with Ubuntu's for whatever reason.

Anyways, QGC is compiled and runs fine now, so I'm going to close this issue if its fine with you?

thomasgubler, 2015-01-21:
Yes can you maybe at some point add a section to the readme about installing from the ppa on ubuntu 14.04/Mint 17.* ? ;-)

jayceelock, 2015-01-21:
Done and done. Created a pull request https://github.com/mavlink/qgroundcontrol/pull/1179

closed 2015-01-21 by jayceelock