mavlink/qgroundcontrol #1182: File extensions need consistency.

dogmaphobic, 2015-01-31:
Unlike something such as a text editor, where you want full freedom to define file extensions, most, if not all extensions used by files written and read by QGroundControl are very specific. The extensions used to read and write these files should be consistent and enforced where applicable.

This is primarily an issue when creating files. The function used to request a filename from the user should have an option to define if the extension is mandatory (strict), in which case the function will make sure only valid extension(s) are used.

Susurrus, 2015-01-31:
@thomasgubler I'd like to pull you into this issue. #1173 brought up the important point of whether we should allow users to rename file extensions, as most of them are very specific.

We deal with the following extensions for user-created files:
 * .mavlink - MAVLink log files. Binary & timestamped.
 * .qgw - Custom tool widget created by the user.
 * .log - Log files saved the LineChartWidget.
 * .txt - Waypoint file -  Text, possibly tab-delimited.
 * .svg, .pdf - Image files from the QGCDataPlot2D **unsure what this is**.
 * .csv, .txt - Data from the QGCDataPlot2D widget
 * .txt - Parameter file

I'd like to make the proposal to use only a single file extension per feature above (except the image saving format) and make it mandatory to use these proper extensions, unlike how it's done now.

I'd like to propose the following strict file extension naming scheme:
 * .mavlink - No change
 * .qgw - No change
 * .log - Log files saved the LineChartWidget **possibly rename this to CSV or TSV depending**
 * .txt (Waypoint file) - **Rename to .waypoints**
 * .svg, .pdf - No change
 * .csv, .txt (QGCDataPlot2D output) - **Make only .csv**
 * .txt (Parameter file) - **Rename to .params**

I'd really prefer a .csv format for the waypoints and parameters files, but I don't think that's worth doing at this stage. At least doing it this way will make things more consistent and easier to differentiate the different output files QGC can generate.

Susurrus, 2015-01-31:
@LorenzMeier Since this is a big user-facing change, can you weigh in here?

dogmaphobic, 2015-01-31:
I will start working with the plumbing. Adding a flag to the openFileSave telling it the extension is strict and within the function, enforcing the extension. Then it’s just a matter of figuring out the places where this should be enforced and what extensions to use in those cases.

thomasgubler, 2015-02-01:
@Susurrus I am fine with the file ending changes. They seem reasonable. I suggest that for compatibility we still allow all file endings in the file open dialogs, agree?

Susurrus, 2015-02-01:
Yes. But do we want to either warn the user that they're using an outdated file ending and suggest they change it and/or provide a way to change it for them?

Susurrus, 2015-02-01:
And still need to nail down the save format of the LogWidget. The simulator crashes for me and I don't have my hardware. Does it produce CSV or TSV file in reality and just name it .log?

DonLakeFlyer, 2015-02-02:
I agree on not forcing strict extension but default appropriately. But warning on not using default extension seems reasonable as well. OS X does something like this in various places: Are you sure you want to change the extension to ".foo"? [Keep .foo] [Cancel]. Or something like that.

dogmaphobic, 2015-02-02:
> I suggest that for compatibility we still allow all file endings in the file open dialogs, agree?

Yes, this issue is only pertinent to *saving* or *creating* files. We want to always use some strict extension (where applicable).

For opening files, it would be a matter of allowing these *legacy* extensions to be selectable in the file open dialog. This is done by using filters. For instances, the log files are now to use "*.mavlink" for the extension. If and when loading these files, we set the filter to:

```c
"Flight Data Log (*.mavlink);;All Files (*.*)"
```

The filter in the file open dialog defines what files can be selected. Those not matching the filter extension are greyed out and are not selectable. By having the above set of filters, the user can switch to "All Files" and all files become selectable.

DonLakeFlyer, 2015-02-02:
Sounds fine to me then.

dogmaphobic, 2015-02-02:
> I agree on not forcing strict extension but default appropriately. But warning on not using default extension seems reasonable as well.

There are two issues within this issue. The first is the strict enforcement of file extensions for specific file types and cases. The other is what extensions are to be used throughout as suggested by @Susurrus above.

Regarding the addition of a *strict* flag to the file open dialog, my idea of *strict* was originally to enforce the use of a specific file extension (where applicable) and not allow the user to do otherwise. The code so far looks at the file entered by the user and sees if the extension matches those defined by the filter. If it doesn't, a message box is brought up telling the user they've entered an invalid file extension (asking to retry or cancel). If no *strict* flag is set, nothing is checked and the function simply returns whatever the user entered (what the code does today).

Is the warning you suggest above something that should be displayed whenever the strict flag is **not** set and the user enters something not in the filter? In other words, what are the expected behaviors?

Susurrus, 2015-02-02:
There are actually 3 issues:
 * What are the file extensions we SHOULD use?
 * How to enforce strict file naming in saving files?
 * What do we do about loading files?

I believe the first one is pretty much addressed, though still need to determine if we can use tsv/csv for the LineChartWidget.

The second is to not allow the user to select a different file ending. For example if the user is saving a .mavlink file and they type in the name "my_log.mavlink", the resultant file should have that exact name. But if they type in **any** other name, it should just have ".mavlink" appended on the end of it. No need to alert the user or do anything. Just silently correct it for them.

For the 3rd, I like the suggestion of having 2 file filters: one for the actual file name we're expecting and another that allows all files (like `"Flight Data Log (*.mavlink);;All Files (*.*)"`). The All Files filter is only for legacy support and should be removed eventually.

DonLakeFlyer, 2015-02-02:
All fine by me. 

Only slight disagreement is that I think the "All files (*.*)" should always be there. For my own use I tend to do things like rename foo.param files to things like foo.param.backup for saving. Would be nice to be able to get to these without having to rename back.

dogmaphobic, 2015-02-02:
I also agree with leaving "All Files (*)" there for good. I see no reason to prevent the user from entering arbitrary file names. When they switch to "All Files", it implies they know what they are doing. The consumer of the file has to make sure it can handle what is coming any way. If it can't, it will then tell the user they are trying to loading something that the code can't handle.

dogmaphobic, 2015-02-02:
> But if they type in any other name, it should just have ".mavlink" appended on the end of it. No need to alert the user or do anything. Just silently correct it for them.

Not quite so silent. The code will have to check and see if the (silently renamed) file already exists. If it does, it has to ask the user to confirm overwriting it.

If this is the agreed behavior, I will change the code (displaying the "Invalid Extension" dialog) to the behavior described above by @Susurrus.

DonLakeFlyer, 2015-02-02:
Makes sense

Susurrus, 2015-02-02:
Yes, as long as the dialog is only shown in the case of a file conflict.

And @DonLakeFlyer, I would consider the *proper* way to save files like you desired would be to name them `foo.backup.param`. Because the .param is used to indicate file type and not be a part of the name, so it should stay the same regardless of what you want to call the file. But this is just an anecdote and I'm fine continuing to allow this usage.

dogmaphobic, 2015-02-03:
This is all done and ready. I'm just waiting for my other 2 Pull Requests to go through to avoid merging issues as this has been a parallel branch (still not 100% comfortable with GitHub).

dogmaphobic, 2015-02-03:
> Yes, as long as the dialog is only shown in the case of a file conflict.

Yes, the dialog is shown when a (silently renamed) file turns out to exist and the user must be warned. When selecting a file within the file dialog, the dialog itself (Qt) handles the file overwrite question. In this case, we are renaming a file after the fact and as such, we have to ask the user ourselves.

With that said, it makes me nervous to rename the file (adding an extension) behind the user's back.

DonLakeFlyer, 2015-02-03:
The extension adding thing is not a completely uncommon practice. If we get grief from users we can adjust later given feedback.

dogmaphobic, 2015-02-03:
No problem. Merging you last couple of merges so I can wrap this up...

dogmaphobic, 2016-01-13:
This has been done for almost a year. Just forgot to close it.

closed 2016-01-13 by dogmaphobic