mavlink/qgroundcontrol #1185: Remove *selectedFilter* argument from File Dialogs

dogmaphobic, 2015-02-02:
Given that it's not used, removing it from the QGXFileDialog file (static) functions. If you need to know what file type was selected, you can always:

```c++
QFileInfo fi(filename);
QString fileExtension(fi.suffix());
if (fileExtension == QString("foo")) {
    // do something
}
```

dogmaphobic, 2015-02-02:
Bah! Created an issue as a (hard to die) habit. This could have gone straight to a PR...

Susurrus, 2015-02-02:
Please put the text from your comment in your git commit. It'd be nice to have that info logged in the git history.

dogmaphobic, 2015-02-02:
> Please put the text from your comment in your git commit. It'd be nice to have that info logged in the git history.

Which one ? :smile:

closed 2015-02-03 by dogmaphobic