mavlink/qgroundcontrol #1189: Move QGCDataPlot2D into a separate tool

Susurrus, 2015-02-04:
Now hear me out, as there are a few reasons for this:
 1. The QGCDataPlot2D is hard to discover for new users. It's only available when QGC is first started. As soon as you connect to a UAS, it cannot be accessed again. I imagine many users wonder where it went. From the code it looks like the `QGCDataPlot2D` is replaced with the `Linecharts` widget on connection of a UAS (`src/ui/MainWindow.cc:1225`)
 2. The QGCDataPlot2D has a complicated UI with no guidance for the user. This is a usability nightmare to expose to new users (think of someone who just bought the Pixhawk but has minimal engineering backround otherwise).
 3. It is only usable for offline data analysis, so making it a stand-alone tool wouldn't change the workflow. In other words, the current workflow for it is to connect to a UAS, save a CSV file from the Linechart widget. Open a new QGC instance (possibly closing the old one) and load that file for analysis. Since you're already opening a new QGC, it'll be less resource intensive to just have this stand alone tool running alongside QGC instead of multiple QGC instances.
 4. It's a large amount of code that has very little user-facing code (since it's only available before connecting to a UAS) and so is likely to have plenty of bugs in it. If people are able to run it stand-alone, it'll likely get more use since it will always be available. Additionally, if it crashes, it doesn't make people think any less of the main QGC program, since it won't affect their ability to operate their UAS.

DonLakeFlyer, 2015-02-04:
I never quite understood #1 myself. Just realized a little while ago that it only showed up when not connected! My suggestion would be to rework this along with a top-level menu item re-work I have in my head. This goes with the whole Setup/Plan/Fly/Analyze top level button change which at this point is just re-wording. But long term I'd like Analyze to really be "Analyze". I could see one of the tools available there being the line chart widget, available all of the time. No association with connected or not. Imagine Analyze taking you to a screen similar to the new Setup View I just pushed with tool buttons along the left hand side. I think that could work.

Bit concerned about adding a separate tool/window/instance. We've stayed away from that so far. It's a lot of additional complexity in a lot of areas (build, base architecture to support differing instances, ...).

Susurrus, 2015-02-04:
The thing with analysis is that you don't need it to be heavily integrated to QGC to be useful. If this analysis tool works on generic CSV data, it's a disservice to burden it with its QGC legacy. It'd be great if it was a standalone tool for CSV analysis. I'd use it for data that I generate outside of QGC. Other people could use it too. Bundling it with QGC just limits its audience versus separating it increases it. And with open source projects, the network affect is really important to take advantage of.

Susurrus, 2015-03-10:
Feature request that I won't be following up on so I'm closing it.

closed 2015-03-10 by Susurrus