mavlink/qgroundcontrol #1194: Replace Connect button with a link dropdown

Susurrus, 2015-02-04:
This button doesn't make sense with multiple connections. It also doesn't track the state of a connection. Instead this UI element should be a dropdown listing all of the Links that are available.

DonLakeFlyer, 2015-02-04:
When there are multiple Links connected the toolbar button should switch to Disconnect and be a dropdown of links to select which one to disconnect. Is something not working or do you want something else?

Susurrus, 2015-02-04:
Sorry I don't think I was clear. I meant this should apply to the Communications Console widget. I rewrote the issue text and left that part out. Ooops.

Susurrus, 2015-02-04:
But this does bring up the issue of what does the Disconnect button mean on the main toolbar when multiple links are active? I'm not sure what will happen from the code and the UI doesn't allude to it either. But maybe this is a new issue.

Susurrus, 2015-02-04:
Whoops, there already is a dropdown there. In this case I suggest the connect button be removed from the communications console. With the toolbar item active and always visible, it doesn't make sense.

DonLakeFlyer, 2015-02-04:
Main toolbar Disconnect button does this:

"When there are multiple Links connected the toolbar button should switch to Disconnect and be a dropdown of links to select which one to disconnect. "

Communications Console connect thing does seem whacky and doesn't track connected correctly.

Susurrus, 2015-02-04:
> Communications Console connect thing does seem whacky and doesn't track connected correctly.

Then you fine with just removing it?

DonLakeFlyer, 2015-02-04:
Most probably. Have to look to see if it is doing anything else with it.

closed 2015-10-25 by DonLakeFlyer