mavlink/qgroundcontrol #1198: Windows build has warnings/errors in debug post-build steps

Susurrus, 2015-02-04:
I get the "A subdirectory or file release\platforms already exists. Looks like there is a mkdir command in the output build. It should only make the directory if it doesn't exist.

I also get the error "The system cannot find the path specified. 0 file(s) copied". I think this is in reference to the `platforms` folder not existing in the output build directory. I think instead of `mkdir release\\platforms` it should be `mkdir \"$$DESTDIR_WIN\\platforms\"`.

Susurrus, 2015-02-04:
But even without this "platforms" directory, things run fine for me on Windows 8.1

DonLakeFlyer, 2015-02-04:
Can you add this to that issue which says to use the new win deploy tool? I think that will fix this as well.

Susurrus, 2015-02-04:
This isn't for deployment, this is just to build the project in debugging mode. Does that affect this somehow?

DonLakeFlyer, 2015-02-05:
The work that happens in QGCSetup.pri happens for all builds and is a bit of a mix of things needed for install which aren't really needed if you already have Qt installed. With the windeployqt thing, the platforms part of the Windows specific work in QGCSetup.pri would go away. The install target is more of just a setup tool creation step which packages the existing directory structure. The actual packaging happens in QGCSetup.pri which is done for all build types. The reason for this is to keep the way builds that you build yourself as similar as possible to the bits and structure for builds that you install. The idea is to keep the bugs down to a minimum that would only occur when you build and installer version.

closed 2015-10-25 by DonLakeFlyer