mavlink/qgroundcontrol #1200: Looking for any solution to use more than 2 ardrone2 in same time  QGC

nimaaghli, 2015-02-05:
Hello everybody,
I was surprised when i realized no one used QGC to control multiple ARdrones.
As you know there is no parameters in list of Ardrone parameters regarding ID to be distigushed by QGC like MAV_SYS_ID all we have is : 
BATTERY-IS-HD 0.000000
CAM-RECORD_HORI 1.000000
CAM-RECORD_VERT 0.000000
DEF-VALID-RAD 3.000000
DEF-VALID-TIME 0.001000
DELAY-COM-LOST 0.000000
DELAY-VIDEOSTOP 10.000000
FLIGHT-ZONE-RAD 1000.000000
HEADING-AUTO 1.000000
RET-HOME-ALTI 0.000000
SPEED-M-S 2.000000
WPT-ALTI-MAX 3.000000
WPT-ALTI-MIN 1.000000

I really Need to list drones in software . Ardrone forum users has no idea how to solve the issue . My question is that it possible to change and modify source code in a way that the code gets the numbers from one of available parameters i have in ardrone ? for example in order to distinguish MAV_SYS_ID it uses BATTERY-IS-HD then i can set numbers like BATTERY-IS-HD =1,BATTERY-IS-HD=2 and so on?

DonLakeFlyer, 2015-02-05:
The system ID will be stamped on each mavlink message that an ardrone sends. This id stamp is then used to distinguish which uas is sending the message so that QGroundControl can do the right thing. In the case of an ardrone I believe all mavlink messages from multiple drones would be coming over the same communication link. Because of that, there is no way to figure out which drone the message is coming from. They will all look to be coming from the same place. I don’t think it’s hackable.

nimaaghli, 2015-02-14:
@DonLakeFlyer i have 2 ardrone they are in same ad-hoc network one has .1 and .2 ip adress if i use 2 computers joined to ad-hoc network is it possible to do something ? i have to have them in ad-hoc network 
there is something mentiond in here for http://www.ugcs.com/downloads/v_1_2/Manual%20ardrone%201.2.pdf can i do same with QGC

DonLakeFlyer, 2015-02-15:
Seems like you might be able to do the same thing with two QGC's on two computers.

DonLakeFlyer, 2015-02-25:
Closing since Issue are for bugs.

closed 2015-02-25 by DonLakeFlyer