mavlink/qgroundcontrol #1201: Support for MAV_PARAM_TYPE_UINT16/INT16 incomplete

DonLakeFlyer, 2015-02-05:
On Feb 5, 2015, at 9:38 AM, Donald Gagne <don@thegagnes.com> wrote:

In looking at the code although MAV_PARAM_TYPE_UINT16/INT16 are defined in the mavlink headers they are not fully implemented through the QGC codebase. If you stick the 32 bits you should be ok.

On Feb 5, 2015, at 12:34 AM, Xiang Gao <gx.feyu@gmail.com> wrote:

Hi, all.
I have encountered a problem using QGroundControl for my self-designed flightcontroller.
I try to send all my paremeters to QGroundControl using Mavlink 1.0 #22 param_value.
It works fine when the param.type equals 
MAV_PARAM_TYPE_UINT32, 
MAV_PARAM_TYPE_INT32, 
MAV_PARAM_TYPE_INT16

But when it comes to  MAV_PARAM_TYPE_UINT16, the QGroundControl just cannot recognized the parameter and puts:
"[!] at src\uas\UAS.cc:2494 - "INVALID DATA TYPE USED AS PARAMETER VALUE:  3"

Does anyone know whether the QGC supports the MAV_PARAM_TYPE_UINT16 type parameter? Since on  http://qgroundcontrol.org/mavlink/parameter_protocol, it says:

Supported Data Types
MAVLink v1.0 supports these data types:

uint32_t - 32bit unsigned integer (use the ENUM value MAV_PARAM_TYPE_UINT32)
int32_t - 32bit signed integer (use the ENUM value MAV_PARAM_TYPE_INT32)
float - IEEE754 single precision floating point number (use the ENUM value MAV_PARAM_TYPE_FLOAT)

If it's true, why can I still use MAV_PARAM_TYPE_INT16?

Susurrus, 2015-02-05:
Note that everything besides UINT32/INT32/FLOAT aren't implemented, so really this issue applies to uint8/int8/uint16/int16 types.

Susurrus, 2015-02-05:
I've hacked at this before, and while it's easy cause it's a lot of cut/paste, there's tons of duplicated code for passing parameters around and it ends up changing a ton of things.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier