mavlink/qgroundcontrol #1207: Black setup window on QGC

TSC21, 2015-02-08:
Hi there,

I'm trying to download the latest firmware to Pixhawk using the QGC GUI, but strangely I can't get the "Setup" window to show anything:
[image]
What can be the problem? One thing I notice is that when I change to the Setup window, I get this in the terminal:
```bash
QQuickWidget: render scenegraph with no context
```
May this be QT related?

Thanks in advance!

TSC21, 2015-02-08:
Update:

I tried to clean all (which included clone the repo again and compile).
Now, pressing the setup button gives me the following:
```bash
QObject::connect: Cannot connect (null)::summaryButtonClicked() to SetupView::_summaryButtonClicked()
Segmentation fault (core dumped)
```
Any idea?

DonLakeFlyer, 2015-02-08:
If you switch to Qt 5.4 as show in this Pull: https://github.com/mavlink/qgroundcontrol/pull/1205 <https://github.com/mavlink/qgroundcontrol/pull/1205>.  It will all work.

TSC21, 2015-02-08:
Yes solved the problem. For the record:
1. add `ppa:beineri/opt-qt54-trusty` to source list:
```bash
sudo add-apt-repository ppa:beineri/opt-qt54-trusty
2. install `qt54` and its apps (p.e.: `sudo apt-get update && sudo apt-get install qt54tools qt54base qt54declarative qt54serialport qt54svg qt54webkit qt54quickcontrols qt54xmlpatterns qt54x11extras qt54websockets qt54sensors qt54script qt54quick1 qt54qbs qt54multimedia qt54location qt54imageformats qt54graphicaleffects qt54creator qt54connectivity qt54canvas3d`)
3. add `source /opt/qt54/bin/qt54-env.sh` to `~/.profile`.

Probably better to add this to https://github.com/mavlink/qgroundcontrol readme, right?

DonLakeFlyer, 2015-02-11:
I"m re-opening this because there is still an issue on Qt5.3 with a black setup button list after connect. I can finally repro myself and will try to fix on 5.3 before we just move to 5.4.

DonLakeFlyer, 2015-02-11:
Looks like it has nothing to do with the qml we are using. It appears bug in the QQuickWidget implementation in 5.3.2. Works fine in 5.3.0 and 5.4, but not 5.3.2.

DonLakeFlyer, 2015-02-11:
Verified that Qt 5.4 fixes the problem. I'm going to see what happens if I switch all OS to Qt 5.4.

Susurrus, 2015-02-11:
Is a fix for this pending for Qt 5.3.3? I'm really wary of us simply doing a large jump in Qt versions whenever we hit a bug. Every jump we've done we've found a bug that's pushed us further forward, and many older distros don't have these newest versions in their repos and makes compiling on Linux much more difficult.

I'd really like us to stay 1 major version behind of Qt if possible.

DonLakeFlyer, 2015-02-11:
No idea if a fix is pending. We've been moving forward to get new features. This is the first time that I know of we moved forward to work around a bug. I’ll verify whether 5.4 works on all OS and then we’ll move forward. We are already past the point of having to deal with distros not having the latest Qt version. I see no reason to stay behind in releases.

@thomasgubler Any issues with moving archlinux to 5.4?

Susurrus, 2015-02-11:
Most distros within the last year have 5.3, so I wouldn't say we're past it. I really suggest against this move before figuring out what upstream's plans are. Someone should really look into it. Qt is a big project with customers and they'll probably have a backported fix pending for 5.3 if someone cares to look.

DonLakeFlyer, 2015-02-11:
@thomasgubler Is putting Qt 5.4 an issue on archlinux?

thomasgubler, 2015-02-12:
> @thomasgubler Is putting Qt 5.4 an issue on archlinux?

arch linux is on 5.4

DonLakeFlyer, 2015-02-12:
@thomasgubler Can you try https://github.com/DonLakeFlyer/qgroundcontrol/tree/Qt5.4 on archlinux? Unit tests pass on OS X, Windows and Ubuntu 14.04. I'm doing deeper testing on these OS now.

thomasgubler, 2015-02-12:
@DonLakeFlyer looks good. One minor comment and  I am not sure if this is the same on the master branch: the box with the vehicles should extend over the whole screen if there is enough space:
[image]

DonLakeFlyer, 2015-02-12:
Same on master. Can you put in an issue?

thomasgubler, 2015-02-12:
Issue created here: https://github.com/mavlink/qgroundcontrol/issues/1231

DonLakeFlyer, 2015-02-12:
@Susurrus are you on Fedora? Can you try the branch with Fedora?

Susurrus, 2015-02-12:
What am I testing? Just the build process on Fedora? Looks like all versions have 5.4.0.

I ask because I don't have a PX4, so I don't think I can test the setup configuration stuff you guys are working on.

DonLakeFlyer, 2015-02-12:
Build and quick check/connect through of top level items to make sure there isn’t something strange with 5.4 and your stuff.

Susurrus, 2015-02-12:
What branch am I testing, `jgoppert:qt-readme-5.4`?

DonLakeFlyer, 2015-02-12:
Branch is  https://github.com/DonLakeFlyer/qgroundcontrol/tree/Qt5.4

Susurrus, 2015-02-12:
```
In file included from src/uas/../../libs/eigen/Eigen/Dense:6:0,
                 from src/uas/../../libs/eigen/Eigen/Eigen:1,
                 from src/uas/UASManagerInterface.h:38,
                 from src/uas/UASManager.h:34,
                 from src/uas/UAS.cc:23:
src/uas/../../libs/eigen/Eigen/Geometry:33:39: fatal error: src/uas/../../libs/eigen/Eigen/src/Geometry/OrthoMethods.h: Permission denied
 #include "src/Geometry/OrthoMethods.h"
                                       ^
compilation terminated.
```

There's no `src/Geometry` directory. Dunno what's going on here.

dogmaphobic, 2015-02-12:
Yes it does. It seems the include path is wrong though...

[image]

DonLakeFlyer, 2015-02-12:
Strange. Wonder why this ever worked? Is this in the openscenegraph code path? I can't build that on my Ubuntu VM.

Susurrus, 2015-02-12:
Note that that compile was using just the common messageset, no specific dialect. I'm now recompiling with the ardupilotmega dataset. Here's my output of qmake-qt5:
```
Project MESSAGE: Qt version 5.4.0
Project MESSAGE: Linux build
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /home/bwmairs/Projects/qgroundcontrol DESTDIR /home/bwmairs/Projects/qgroundcontrol/release TARGET qgroundcontrol
Project MESSAGE: Using MAVLink dialect 'ardupilotmega' specified in user_config.pri
Project WARNING: Skipping support for OpenSceneGraph (missing libraries, see README)
Project MESSAGE: Skipping support for Google Earth due to Issue 1157
Project WARNING: Skipping support for XBee API (missing libraries, see README)
Project WARNING: Skipping support for 3DConnexion mice (missing libraries, see README)
Project MESSAGE: Skipping support for RT-LAB (unsupported platform)
Project WARNING: Skipping support for speech output (missing libraries, see README)
```

DonLakeFlyer, 2015-02-12:
ardupilotmega is no longer supported, only common. Now that I think of it we've run into the .. usage on include files problem. Has to do with the compiler version more than Qt I think. I'll fix it up.

Susurrus, 2015-02-13:
Alright, building it worked on ardupilotmega. Looks like something's wrong with building with just the common message set (just don't set MAVLINK_DIALECT in user_config.pri). Can someone else reproduce this/

DonLakeFlyer, 2015-02-13:
The include path is wrong. I'll fix it

DonLakeFlyer, 2015-02-25:
Fixed with move to 5.4

closed 2015-02-25 by DonLakeFlyer