mavlink/qgroundcontrol #1208: Calibration indicators not turning green

jgoppert, 2015-02-08:
The color of the mag/gyro/accel calibration indicators now don't change to green after calibration on master.

DonLakeFlyer, 2015-02-09:
Looks like in latest px4fmu firmware master the parameters have changed for calibration. This is also causing the problems with the black screens you are getting because parameters are missing that are expected.

@LorenzMeier Did something change with parameters associated with calibration?

DonLakeFlyer, 2015-02-09:
Looks like they may have changed to CAL_* parameters.

LorenzMeier, 2015-02-10:
Hi Don,

I didn't realize that the black screens can be associated to that. Yes, the param names had to be adjusted once to the CAL_ set.

DonLakeFlyer, 2015-02-10:
@LorenzMeier Should the calibration code just be changed the CAL_...1_ version of these new params?

LorenzMeier, 2015-02-10:
I have a PR somewhere. Let me dig.

DonLakeFlyer, 2015-02-10:
I can fix it. I'm going to need to change the VehicleComponent code as well to implement an interim parameter mapping capability. So it's going to get pretty involved to support both Stable and master since they use different parameter sets.

LorenzMeier, 2015-02-10:
Here you go.
https://github.com/mavlink/qgroundcontrol/pull/1221

LorenzMeier, 2015-02-10:
Its not too involved actually. If you can query if a parameter exists in the set its pretty simple. Since QGC only uses them to check if something has been set up, its not very involved in their actual values.

DonLakeFlyer, 2015-02-10:
It will end up being easier to leave to code connected to the old params for now and re-route in the param manager. Long term it will need to get fancier due to the FactSystem stuff.

LorenzMeier, 2015-02-11:
Fixed in #1223.

closed 2015-02-11 by LorenzMeier