mavlink/qgroundcontrol #1214: Disconnecting a UAS from a live link triggers log file save dialog

Susurrus, 2015-02-09:
I think this has to do with the disconnect/reconnect behavior, as I wouldn't imagine this is intentional. If you get to the limit of your radio range you might go in and out of reception. Having this **modal** dialog pop up several times would definitely limit QGC's usefulness then.

DonLakeFlyer, 2015-02-09:
You didn't disconnect using QGC right? You just broke the connection from a hardware standpoint? If so, the broken link code needs to be modified to look different than a user-driven disconnect/reconnect. Should be pretty straightforward. Dialog needs to be modal so it doesn't lose the system state out from under it. For example a re-connect would cause loss of the temp log file.

Susurrus, 2015-02-09:
Why doesn't the temp log system just save temp logs everytime it connects using an incrementing style. Then when QGC is launched again it can clean any extra ones up. This would be more complicated, but I'm very wary of modal dialogs in a system that's supposed to control robotic systems.

DonLakeFlyer, 2015-02-09:
I thought there was code which automatically disconnected/reconnected when it lost a heartbeat for some length. There doesn't seem to be. The only way the link would be disconnected would be if the user specifically disconnected the link. If you just lose communication over the link when you go out of range for example the actual link will stay connected as far as I can see. The toolbar might say connection lost but that's just because it lost heartbeat. What's the problem here? Or is there some other repro?

Susurrus, 2015-02-09:
Oh there definitely is. I've been debugging stuff and it's the most annoying thing ever because I have to check whether QGC is connected or not. It continually connects/disconnects and never gives up.

But I don't understand what you're asking about.

DonLakeFlyer, 2015-02-09:
Ok. I need to dig more to find it. The repro case would be to connect QGC over radios and then pull the radio on the non-QGC side?

Susurrus, 2015-02-09:
Yeah, easiest is to stop the autopilot transmitting: hold it in reset, power it off, disconnect the UART, etc.

DonLakeFlyer, 2015-02-09:
Found it. It's a "feature" of the SeriaLink code only. Will need to distinguish between user disconnected and SerialLink code trying to do it's thing.

Susurrus, 2015-02-09:
Well, should this be something that's pushed up into the `LinkInterface` code, or possibly the SerialInterface code? Though I don't know what problem this code is trying to solve. Does disconnecting and reconnecting ever fix anything?

DonLakeFlyer, 2015-02-09:
"Does disconnecting and reconnecting ever fix anything?"

That is the important question to get answered. This code may actually cause more problems than it fixes.

@LorenzMeier Commit say: Comms: Re-establish the link reliably if just cycling through bootloader. Any ideas if this code still makes sense? It is only in the SerialLink. No other link types try to do this although they could suffer from the same inability to write bytes when something goes wrong.

DonLakeFlyer, 2015-02-09:
Also, this is the reason that pulling a USB connection ends up with a disconnected link instead of the link hanging around with communication failing on it. Lorenz this is per our gitter discussion. If this code wasn't there, communication links would stay active but with errors. Hence only way the disconnect is from user intervention and we could fix this whole disconnect a link for real removes the UAS stuff.

DonLakeFlyer, 2015-02-09:
This code was working around the fact that on the Pixhawk even after the bootloader times out it still takes a second or so for the USB driver to some up. So even though you plug in and see the port show up as available you can't open it until all that finishes. Hardware connect already has code which waits until the bootloader goes by. But it should also have code to retry the opens with a wait a fews time so that it gives enough time for the USB driver to go live. I just removed the on the fly connect/reconnect stuff and implemented the open retries and you can connect fine ever if you click connect right after the port shows up.

With that the question comes down to automatic disconnect/reconnect in link errors. Too me this seems bad. If communication is lost you'll lose the heartbeat and QGC will still warn you. Also this disconnect/reconnect code only keeps trying for a few seconds. So if something goes from while you are away from the machine, it will try for a few seconds and then leave the link suddenly disconnected. if comms are restored after that your still dead in the water. To me it makes for sense to leave link up. if it comes back, great it will always just start working again no matter how long it was gone. If the user sees it they are notified that the connection is lost and they can decide whether they want to disconnect or not.

DonLakeFlyer, 2015-02-25:
This is fixed now. You will only get log save when the user manually disconnects the last link. Loss of connection will not cause it to pop up.

closed 2015-02-25 by DonLakeFlyer