mavlink/qgroundcontrol #1217: messy code, terminal, nsh

chzw517, 2015-02-10:
[image]

when i press the connect button it print out endless messy code... are there any one confront the same issue? how to solve?

dogmaphobic, 2015-02-10:
@DonLakeFlyer It reminds of a conversation about "Comm Terminal" visible to new users... :)

DonLakeFlyer, 2015-02-10:
There is something going wrong with the terminal code in that it isn't switching over to the nsh shell. What  you are seeing are the mavlink messages going by. Terminal code should push out three carriage returns which should switch the link from mavlink to the shell.

LorenzMeier, 2015-02-11:
The terminal and the communication console are duplicates of each other. I expect the console to be a bit more flexible and would suggest to keep it and remove the terminal. But its up to you.

closed 2015-02-11 by LorenzMeier