mavlink/qgroundcontrol #1218: FactSystem\FactLoader.cc:201 - Invalid variant type QChar

Susurrus, 2015-02-10:
Looks like support for Chars was broken when moving to the new factsystem. Seems to load these variables just fine, however, but the warning is still worrisome.

DonLakeFlyer, 2015-02-10:
This is a single character parameter?

Susurrus, 2015-02-10:
I have them set as `PARAMETERS_DATATYPE_UINT8` in the MAVLink parameter protocol

DonLakeFlyer, 2015-02-10:
Ok. Now i got it.

Susurrus, 2015-02-13:
And what's weird is the debug message to the console when receiving a parameter of type UINT8:
```
UASLog: Received PARAM_VALUE "ModeAuto" QVariant(QChar, '
FactLoaderLog: Adding new fact "ModeAuto"
FactLoaderLog: Updating fact value "ModeAuto" QVariant(QChar, '
"ModeAuto" Invalid variant type QVariant::QChar
```

And no, those aren't typos above, it seems QChars are supposed to be ASCII, which my values aren't because they're actually bools (0 and 1), so displaying either of those just breaks that line of console output.

And weirdly enough things do work in QGC, but it reports an error when setting the value for all UINT8 datatypes. I think this is the same issue as #1201.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier