mavlink/qgroundcontrol #1224: continuous error messages on console with HILSIM

kd0aij, 2015-02-10:
HILsim with Xplane9 on Ubuntu 14.10 64 bit. Getting continuous messages from QGC:

src/comm/QGCXPlaneLink.cc556 UDP datagram overflow, allowed to read less bytes than datagram size

dogmaphobic, 2015-02-11:
I can fix this. The code sets an arbitrary (low) limit and doesn't seem to loop to get the remaining data.

kd0aij, 2015-02-11:
Would this corrupt the simulation?

dogmaphobic, 2015-02-11:
Sorry, I haven't had a chance to look into the details yet. Yes, it appears it is only reading the first 1,000 bytes of a datagram and discarding the rest. On the other hand, it could be that the remaining bytes will be read the next time data arrives (when that event is triggered). If that's the case, the issue would be of latency (and losing a few bytes at the end of a connection).

kd0aij, 2015-02-11:
Thanks, I appreciate you checking into this.

DonLakeFlyer, 2015-02-11:
The datagram is larger than QGC expected. Which means the communication protocol setup with XPlane in likely incorrect, since it is sending more information in a single packet than QGC expects. I don't think it's a QGC problem, more likely an XPlane protocol setup problem.

DonLakeFlyer, 2015-02-11:
That said, better error handling and error messages here would be useful. FlightGear code does a bit better job of this.

kd0aij, 2015-02-11:
@DonLakeFlyer Thanks, you were correct about this being an Xplane setup problem. It was sending SNAP records which were 2665 bytes long. I must have accidentally checked an extra box in the network setup menu. We could increase maxLength, or just ignore SNAP datagrams. A more specific warning could be issued if this affects performance, but I don't know whether that is the case.

DonLakeFlyer, 2015-02-11:
We'll leave this issue open and I'll take a look at some point. I eventually want to do a full cleanup of HIL stuff.

closed 2015-10-25 by LorenzMeier