mavlink/qgroundcontrol #1247: UASView slow to update

Susurrus, 2015-02-17:
I've recently implemented the throttle output for my vehicle with the VFR_HUD message. It's really useful to know the throttle value when underway. But the UASView is very slow to update, so it gives a much slower-than-realtime status, looks to be about every 3s. This is the same regardless of the low-power mode.

Would people be amenable to moving this to a 1s update rate? That'd give much better feedback to the user.

Susurrus, 2015-02-17:
Looking at the code it looks like the update rate should be .8s normally and be 3x that during low-power mode. So I'm not certain how 3s is occurring in the non-powersave mode.

Susurrus, 2015-02-17:
Digging through this a bit more, it looks like the update logic has gotten quite complex because it's trying to account for the following:
  * Animating a color fade for the heartbeat messages
  * Flashing red at 5Hz when the system is disconnected
  * Responding to Qt widget events, for a reason I don't understand.

The reason I say it's currently too complex is because it doesn't seem to update the whole UI for a UAS at the 800ms intervals, instead updating the color fade quickly but not otherwise updating things like the uptime timer and other stuff.

My proposal here is to remove the color fade completely as well as the other widget event timer setting (which doesn't seem to do anything anyways). We'd keep the 5Hz flashing that is per-UAS but ditch the rest to keep the logic simple.

So, do people like the heartbeat fade is I guess the pending question for this change.

Susurrus, 2015-02-17:
Or maybe it's just worth changing it so that the throttle value is updated everytime a VFR_HUD message is received but otherwise leaving the logic as it stands.

LorenzMeier, 2015-02-17:
The issue is that a lot of other messages arrive at 5-10 Hz, making a widget redraw unacceptable at that rate.

Susurrus, 2015-03-10:
That's fair. Since I won't have time to fix this and it's a feature request, closing for now.

closed 2015-03-10 by Susurrus