mavlink/qgroundcontrol #1261: Incorrect UDP link causing multiple "Warning a second system is using system id (247)" popup boxes

ggregory8, 2015-02-22:
As mentioned on gitter. If I enter incorrect UDP link for target host and it detects its own (QGC) system ID it creates the MAVLink Protocol popup box with message "Warning a second system is using system id (247)". If you do not close this box within ~1s another one will popup over top. They will continuously open. If you manage to quickly click 'ok' on a couple of popups then they will stop opening new popups.

Steps to reproduce:
1. Use master QGC from 2015/02/22 (v2.3.0)
1. Add new Communication Link with this following settings:
Link Type: UDP
Listening Port: 14550
Add Target Host: localhost:14550         i.e (127.0.0.1:14550)
2. Click Connect

Note that the default target port used to be 14555, now it uses 14550 which is the same as the default listening port. Having these two the same causes the duplicate system ID detection.

Using the stable release (v2.2.1) on windows, following the same steps, generates the same popup, but only once as it should.

ggregory8, 2015-02-23:
[image]

DonLakeFlyer, 2015-02-25:
@dogmaphobic Gus can you take a look at this? I think it's just a typo in QGC_UDP_PORT but want to make sure there wasn't a reason for the change. Also probably a good idea it verify the host and listening port are not the same.

dogmaphobic, 2015-02-25:
The original code, before I changed to LinkConfiguration had the default port set to 14550. I just used what I found there:

```c++
    UDPLink(QHostAddress host = QHostAddress::Any, quint16 port = 14550);
```

Should I change the default to 14555? I will take a look at not allowing a host:port conflict.

DonLakeFlyer, 2015-02-25:
Config code did this though:
-    QString hostName = QInputDialog::getText(this, tr("Add a new IP address / hostname to MAVLink"),
-                       tr("Host (hostname:port):"), QLineEdit::Normal,
-                       "localhost:14555", &ok);
-    if (ok && !hostName.isEmpty())
-        link->addHost(hostName);

I'm not familiar with UDP but seem like from the above it should be 14555.

dogmaphobic, 2015-02-25:
Found it. The error is in the Add Host logic, which is using the same port for targets.

dogmaphobic, 2015-02-25:
Yep :)

dogmaphobic, 2015-02-25:
Pushed in #1265

DonLakeFlyer, 2015-02-25:
Fixed

closed 2015-02-25 by DonLakeFlyer

ggregory8, 2015-02-26:
@DonLakeFlyer I have just pulled master and recompiled. The default target port has changed to 14555 which is good, but the multiple popups is still occurring on 'incorrect' settings of target host.

This is occurring during the same procedure I mentioned above i.e UDP target of 127.0.0.1:14550

dogmaphobic, 2015-02-26:
My fix was specific to the wrong default port. It was a safe and isolated fix. The multiple dialog pop ups is a different problem. I don't have anything that connects through UDP to test this.

ggregory8, 2015-02-26:
I can reproduce this problem simply by following the steps in post 1. You don't need any UDP connections. The problem exists if trying to connect to 'yourself'.

DonLakeFlyer, 2015-02-26:
The host and listening port need to verified that they aren't the same.