mavlink/qgroundcontrol #1263: Game controller interface

mkschreder, 2015-02-24:
I want to use my PS2 clone usb game controller with qgroundcontrol to control the copter with two joysticks on it. Is support for bluetooth remote from pc planned any time soon? What other alternative would you recommend for this kind of thing until support is added?

LorenzMeier, 2015-02-24:
Have you actually tested it? Does your operating system support it? Because we support pretty much all joysticks.

CodeChief, 2015-02-24:
How will you deal with the throttle? Guess you have to calibrate upwards movement only else you'll take off accidentally if you let go of the controller stick (dangerous). On the other hand if you want 3D flight capability and have the necessary ESC/firmware and 45 degree props, a game controller would be ideal with the throttle going both up and down.

mkschreder, 2015-02-24:
I got it working. I had to add support for setting manual mode to my fw, then plug in joystick, then restart qground control, then enable joystick in settings, then arm and then it worked. :-)

mkschreder, 2015-02-24:
CodeChief, but there is a setting for it in the dialog to have it only go half way. I'm thinking though to have 50% throttle set so that it makes props spin a little below hover. Also props don't spin if not armed.

LorenzMeier, 2015-02-24:
Are you using PX4 or something else on the autopilot side?

mkschreder, 2015-02-24:
Nah roll my own. I have a firmware I'm building on top of libk (GPL) so I have good hardware support for both avr based and stm32 based boards. But I've never built a copter firmware before so getting it stable in the air is proving to be quite a challenge. Anyway, the current idea is to try and get the controls stable and use mavlink over bluetooth serial to test using a game controller connected to qgroundcontrol. So far so good. :-)

CodeChief, 2015-02-24:
Like that, I think we won't need RC radio in the future. A smartphone, serial uplink radio and controller are much better in many ways. Something like the NVIDIA Shield would be a perfect all-in-one design. Of course latency of serial MAVLink fighting with status messages/parameter updates is a challenge, need priority on control messages. Hopefully when full HD (digital) FPV video links are available at decent prices there will be a new faster transport to utilize. Three different radios mostly analog or digital at 1980s baud rates is just backwards.

mkschreder, 2015-02-24:
Exactly. Imagine a transmitter with mavlink support even. Very doable even with today's cheap radios that have a serial port. Just need a good long range serial link. 

I haven't experienced that much of a problem with latency though even at standard 57k. RC control signals don't need to be transmitted that often either. I'm sending all the other messages at ~6hz and that seems to work well enough and rc signals are sent at the default qgroundcontrol rate.

LorenzMeier, 2015-06-02:
Works in master, I use a Logitech game controller.

closed 2015-06-02 by LorenzMeier