mavlink/qgroundcontrol #1272: Firmware upgrade usability

LorenzMeier, 2015-02-26:
The new UI is quite nice, the upgrade itself seems however still to be quite confusing - I stumbled over these steps:

 - The text says you can plug the board now in the first line - it never tells you to unplug the board before though.
 - If it finds the board connected, it aborts asking to unplug the board first. But nothing in the text or UI does indicate that the user has to click upgrade again. Should the system just continue to try instead? If not, it has to tell him boldly that he has to click again.

In general I'm not sure how much of the unplugging / replugging is a red herring. There are situations where the app will fail to send the board into the boot loader and where the user will need to do the unplug / replug cycle. Under normal circumstances however, replugging is not necessary, as long as QGC does properly close all links it has open itself before attempting an upgrade.

From an usability perspective it would be great to get the automation back and only fall back to the current text instructions when the automation times out.

[image]

DonLakeFlyer, 2015-02-26:
I thought about going back to the previous set of steps where you can start with the board connected. I bailed out because it got a little more complicated due to the fact that the upgrade code uses different serial port code than regular communication (qextserialport versus QSerialPort). Previous code used to just use the same port. Still doable but for now I'd rather clean up the unplug first sequence and then tackle the slightly more complicated already connected way of doing it for later.

LorenzMeier, 2015-02-26:
I'm not suggesting to go back to "steps", I think the current state is much better. All I'm suggesting (long-term) is to make an attempt to reboot the board (by disconnecting it in QGC and issuing the reboot command, which should be still in the boot loader interface source code). Only if it fails instruct the user to cycle the USB connection.

This was the state before you started the refactoring and it worked reasonably reliable until the firmware grew over 1 MB and we hit the silicon errata on the MCU. It makes the process completely stateless from the user-perspective.

DonLakeFlyer, 2015-02-26:
I know. I'm not saying it's not possible, just that it's a little more work than it used to be, hence I'm trying to get the simpler coding mechanism in a working state. Then I can move on to the other setup screens. Then I can come back to this. 

It's a little more hassle, than before since you used to be able to use the same serial port connection and now you can't. If connected already, you need to disconnect the QSerialPort connection, reconnect using qextserialport, do the upgrade thing, then reconnect with QSerialPort. All doable in the current ui.

closed 2015-10-25 by LorenzMeier