mavlink/qgroundcontrol #1276: Removing UAS leaves a useles Connect Link button

Susurrus, 2015-02-26:
After deleting the UAS the Unmanned Systems widget now shows the "Connect Link" button, which doesn't do anything when pressed.

This may change after #1266, but steps to reproduce are:
  1. Boot
  2. Connect to UAS
  2. Switch to Plan where I have the widget
  2. Stop the serial stream on the UAS end
  2. Right click Delete UAS
  2. Quit QGC
  2. Got save flight log dialog, cancel
  2. App quits

dogmaphobic, 2015-03-10:
As far as I can tell, all of those widgets need some serious re-work. Most of them were written eons ago and have never been updated. In addition, there is still some work to be done where QGC keeps looking for heartbeats for closed links, which is related to the above.

Susurrus, 2015-03-10:
The easiest solution to this is to just go back to how it used to work, which was show nothing. At least that way nothing's broken, it's just not ideal.

DonLakeFlyer, 2015-03-10:
That's the plan

closed 2015-10-25 by DonLakeFlyer