mavlink/qgroundcontrol #1291: bad path in ubuntu trusty PPA

kd0aij, 2015-03-01:
W: Failed to fetch http://ppa.launchpad.net/beineri/opt-qt541-trusty/ubuntu/dists/utopic/main/binary-amd64/Packages 404 Not Found

correct path is http://ppa.launchpad.net/beineri/opt-qt541-trusty/ubuntu/dists/trusty/main/binary-amd64/

DonLakeFlyer, 2015-03-01:
Where are you seeing this as wrong. README lists ppa as just `ppa:beineri/opt-qt54-trusty`

kd0aij, 2015-03-01:
I followed the Linux build instructions for Ubuntu;
sudo add-apt-repository ppa:beineri/opt-qt541-trusty
(ppa:beineri/opt-qt54-trusty didn't exist)

Then apt-get update returned a 404 error on a bad URL.
I was able to work around the problem by changing the distribution using
the Ubuntu GUI. Screenshot attached.

kd0aij, 2015-03-01:
BTW, the dist originally showed up as utopic instead of trusty

DonLakeFlyer, 2015-03-01:
Screenshots won't go through unless you put them in from GitHub.

DonLakeFlyer, 2015-03-01:
Mark: I'm not much of a Linux person. I just kind of know enough to make myself dangerous. If you could update the README with instructions that work and put up a pull I sure would appreciate it.

kd0aij, 2015-03-01:
created a PR
I know nothing about PPA's, but I'm sure the owner would know how to fix the URL :)

DonLakeFlyer, 2015-03-01:
Hopefully fixed now

closed 2015-03-01 by DonLakeFlyer