mavlink/qgroundcontrol #1292: APPCRASH when sendMessage(msg)

alexevt, 2015-03-01:
Hi,

I added a new widget window to QGC. I want to recieve and send some data with MAVLink from this form. I can recieve data but while trying to send, I get error "qgroundcontrol.exe has stopped working".

I use the following code (HCStatusWindowView.cc):
void HCStatusWindowView::DVLStatusSelectNew() {
mavlink_message_t msg;
mavlink_msg_param_set_pack(mavlink->getSystemId(), mavlink->getComponentId(), &msg, uasTest->getUASID(), 0, "RC_SPEED", 12, MAV_PARAM_TYPE_UINT16);
uasTest->sendMessage(msg);

While uasTest defined as:
UAS* uasTest;
in HCStatusWindowView.h file

The problem occurs in the last line: uasTest->sendMessage(msg);

Please help me with this issue,
Alex.

DonLakeFlyer, 2015-03-01:
If you are reading and writing parameters you should be using the parameter manager not communicating directly with mavlink. You can get the param manager from the UAS. In the code below _mav is a UASInterface*. 

    QGCUASParamManagerInterface* paramMgr = _mav->getParamManager();
    Q_ASSERT(paramMgr);
    
    paramMgr->setPendingParam(compId, “RC_SPEED", 12);
    
    // Tells the param manager to send pending params and to write them to ROM
    paramMgr->sendPendingParameters(true, true);

alexevt, 2015-03-01:
Thanks for your answer. I tried it but still the same error.

By the way, the code I posted it working well when I run it in the UAS.cc file (I put it for test in one of other functions and I see that "RC_SPEED" was changed to 12).

What can it be?

DonLakeFlyer, 2015-03-01:
Can you move this to theQGroundControl google group discussion and we'll continue there? Issue are for QGroundControl bugs. Since this is your own code, this isn't not a QGroundControl bug.

closed 2015-03-01 by DonLakeFlyer