mavlink/qgroundcontrol #1304: Log replay non-operational

LorenzMeier, 2015-03-03:
Replaying mavlink logs is not working any more. It looks like the reason is that the app is holding off initializing the UI before it gets parameters:
```
silenceTimeout 350
TYPE CHANGED TO: 13
Add new UAS:  1
Unknown message from system: 1 message: 87
compId  0 readsWaiting: 0
compId  50 readsWaiting: 442
compId  0 readsWaiting: 0
compId  50 readsWaiting: 364
compId  0 readsWaiting: 0
compId  50 readsWaiting: 217
compId  0 readsWaiting: 0
compId  50 readsWaiting: 217
compId  0 readsWaiting: 0
compId  50 readsWaiting: 217
compId  0 readsWaiting: 0
compId  50 readsWaiting: 217
compId  0 readsWaiting: 0
compId  50 readsWaiting: 217
compId  0 readsWaiting: 0
compId  50 readsWaiting: 198
compId  0 readsWaiting: 0
compId  50 readsWaiting: 180
```

It would be great to bring this back, as it would help significantly during development of the visuals for the flight operation and its a very intuitive way to do post-flight analysis.

DonLakeFlyer, 2015-03-05:
The log file has to have the parameters played back in order for it to work correctly. With the whole disconnect doesn't really disconnect problem, you can end up saving a log from a second connect in sequence where the log file does not have parameter messages in it. So in turn QGC won't run correctly since it doesn't have the params to work with. If you save a log which has the initial parameter list request in it then it should work correctly. At least it does for me. Is that the case you are hitting?

LorenzMeier, 2015-03-08:
Yes, that's the case. The question is if we really need the params to just replay the flight - I would guess not? Instead of holding off the UI setup based on param stream, wouldn't it make more sense to just hold off the config section? The rest should be encoded in the stream already to work properly.

DonLakeFlyer, 2015-03-08:
> The question is if we really need the params to just replay the flight

I'm not sure. It depends on whether any of the non-setup code needs parameters to work correctly. Haven't really looked there. Also what us actually not working? There isn't much code that waits for the params to trigger. Mostly setup as you say. Not sure why this would prevent the rest of the replay to not work.

Feyu, 2015-03-25:
It's very useful function for me recently. I had a sd card recording all the mavlink binary. It will be fantastic if we can bring this feature back!

DonLakeFlyer, 2015-03-25:
I still don't understand what exactly goes wrong here. The only thing which waits for parameters which did not previously is setup. The rest of the code should be the same.

closed 2015-10-25 by DonLakeFlyer