mavlink/qgroundcontrol #1310: Number of satellites shows 99 in new toolbar

Susurrus, 2015-03-04:
According to `common.xml`, specifying 255 for the number of satellites visible should indicate that there is no value for this field. The toolbar shouldn't show this field if that's the case.

dogmaphobic, 2015-03-04:
Good catch. The value sent containing the number of satellites is a *double*. I found it strange and even stranger than once in a while I would get -1.0 satellites. Should the Satellite control be hidden altogether in that case?

Susurrus, 2015-03-04:
I think it should be. It could also just be shown with an X. But, like Lorenz, I'm using a small screen for QGC so I'd just prefer it to be removed entirely.

LorenzMeier, 2015-03-04:
I would propose to keep the GPS icon at all times as long as we're receiving some GPS data - its a quite important status of the system. I would further propose to turn the satellite icon red if no lock is present and keep it blue if lock is present. If the number of satellites is unknown just don't show a count, else present the count.

@Susurrus would that work with you? If you have no GPS, it would also not show up. I keep myself looking in the field often for signs of GPS lock and good reception, and it would be nice to have this easily available.

dogmaphobic, 2015-03-04:
I would rather show it with an X or something to that effect. As to the screen size, I'm adding settings to the preferences dialog allowing you to select what gets to the tool bar or not. That way, you can disable it as a preference instead.

LorenzMeier, 2015-03-04:
@dogmaphobic X might indicate missing, so if the count is negative (unknown) I think leaving it out might be the better choice. You still have the lock indicator you can use to style the icon according to no lock, 2D lock and 3D lock.

Susurrus, 2015-03-04:
The coloring sounds good, but I suggest green when there's a good fix and red when it gets bad, the current blue I'm seeing doesn't help me understand what's going on. Additionally, if there's no satellite number information, just leave out the number but put it in the tooltip so people understand that the UAS isn't reporting that information.

closed 2015-03-05 by DonLakeFlyer