mavlink/qgroundcontrol #1312: [new toolbar] Voltage value overlaps the battery icon

Susurrus, 2015-03-04:
While "10.48 V" is being displayed for the battery voltage I'm seeing the 1 halfway overlapping the battery icon.

LorenzMeier, 2015-03-04:
@dogmaphobic How about limiting it to the first decimal digit and human-rounding the 2nd? 10.48V is much too detailed, 10.5V is good enough. There is also enough jitter on voltage that those numbers would oscillate quite a bit in a real flight.

Susurrus, 2015-03-04:
That sounds good to me. If you needed more detail you could look directly at the MAVLink message data anyways, so no loss there.

dogmaphobic, 2015-03-04:
> While "10.48 V" is being displayed for the battery voltage I'm seeing the 1 halfway overlapping the battery icon.

That shouldn't happen. The code looks for screen density (dpi) to make sure it's consistent across platforms so things like this don't happen. What are you running and what are the physical screen dimensions?

With that said, I will round to a single decimal point.

Susurrus, 2015-03-04:
I'm seeing this problem with the satellite view as well. But I have a myriad of display problems on Fedora 21 XFCE, so I haven't really reported a lot of them as I think it's a Qt/XFCE problem and not necessary QGC (Like most text is displayed way too dark).

LorenzMeier, 2015-03-04:
@Susurrus I still think you should implement a charge level estimator. You'll confuse the heck out of anybody trying to use your boat with that ancient battery technology. Its a 3-line change and save all the argument about the icon ;-).

Susurrus, 2015-03-04:
@LorenzMeier True, but I'm writing my Ph.D. thesis, not working on making this more usable for others unfortunately. We always launched with it fully charged and it can run for longer than there's daylight. This is a waste of time for me at this stage in the project.

Susurrus, 2015-03-04:
Indeed, note that I just upgraded to Fedora 21 and this is no longer an issue. The number is close to the satellite icon for that button, but I'm closing this issue anyways.

closed 2015-03-04 by Susurrus