mavlink/qgroundcontrol #1313: [new toolbar] Don't display battery icon when HEARTBEAT.battery_remaining is -1

Susurrus, 2015-03-04:
It just wastes space. If you want the fact that there is no battery remaining data, make it a tooltip for the user.

Susurrus, 2015-03-04:
Partial fix for #1312

LorenzMeier, 2015-03-04:
I would still keep it default-colored to relate the presented voltage to battery (else there is just XX V, which is less obvious than when combined with a battery icon). You could custom-configure your screen as @dogmaphobic already hinted if your requirements are higher than mine (I have no more space issues with the current state). All the systems I know estimate the battery remaining anyway, and even you might want to consider adding a simple estimator on your system, since its much more obvious than just a voltage.

Susurrus, 2015-03-04:
Is the voltage value really not intuitive when there isn't any icon? What about changing the icon so that it doesn't have an X in it. That makes me think like something is wrong even when it's perfectly fine according to MAVLink's `common.xml` that my system doesn't report battery remaining.

Aside: I'm not planning on adding an estimator to my system as it's not necessary (I'm rolling with 6 car batteries and my boat doesn't fall from the sky when out of juice).

dogmaphobic, 2015-03-04:
> and my boat doesn't fall from the sky when out of juice

You came to mind when I was putting together the "Fly" button... :) Maybe I should look into the type and change it to "Cruise" instead :)

Susurrus, 2015-03-04:
@dogmaphobic Haha, thanks for thinking of me!

So why not "Operate"? There's also "Launch" or "Deploy", but I don't really like those.

LorenzMeier, 2015-03-04:
Without being ignorant of other communities I think its important to choose intuitive language for what 90% use. If we can find something that generalizes well I'm all for it, I'm just not sure what average Joe thinks "Operate" means.

Susurrus, 2015-03-04:
And that's exactly the right approach, but "Fly" isn't even a good descriptor for it. This view isn't about flying the vehicle, it doesn't give you control of the vehicle, whereas Plan does more of that. I'd actually suggest Cockpit as a replacement term for it. It's much more clear this is supposed to give a HUD-like operating view of the vehicle from the vehicle's perspective.

The only problem with renaming that Cockpit, is that it messes up the whole 1-2-3-4 steps overview you guys are using. To be consistent I think a possibly more-clear labeling would be "Setup", "Mission", "Cockpit", "Analysis". If you use the chevron style I saw in screenshots in the original PR, you still get the clear ordering that you had before, but now the intent of each perspective is much more clear and vehicle-type agnostic.

LorenzMeier, 2015-03-04:
I like both the analysis and the proposals - @dogmaphobic feel free to pick whatever works ;-).

Susurrus, 2015-03-04:
And to amend my previous comment, if the chevron style is used (which I really like BTW), I suggest removing the numbers, but I think that's already been discussed.

dogmaphobic, 2015-03-04:
> "Setup", "Mission", "Cockpit", "Analysis"

More likely would be "Setup", "Mission", "Crash", "Analysis"... 

Jokes aside, "Cockpit" doesn't suggest an action like the others. How about "Run" or "Execute"?

DonLakeFlyer, 2015-03-04:
I would assume that over 90% of the users of QGC will “Fly”. Trying to move down to least common denominator doesn’t make sense to me. Then we end up with someone that nobody understands.

Susurrus, 2015-03-05:
I don't know if a continuation of this conversation is desired, but since we're spitballing here, my argument is that these shouldn't be verbs. They're views of the system. You don't fly the system from the Fly view. It's flying itself, or under direct manual control which is independent of the perspective, and the Fly view is really a "View from the Cockpit" view. I think it's a misnomer to call it Fly. And since these are perspectives, I think the naming I proposed makes more sense. A mission perspective, which is like a general's view of his troops where you can see them and a map and the high-level mission objectives. Then you can go to a cockpit perspective for any one of your drones. Both of these are fine to use for operating your vehicle(s). So while the Setup (maybe Config, or Hanger) and Analysis perspectives don't fit this visual paradigm quite as well, it's just something to think about. And like you guys, I'm just tossing around ideas, I really don't care about the terms.

DonLakeFlyer, 2015-03-05:
The point is that first you Setup your vehicle, then you Plan your flight, then you Fly your flight, then you Analyze your flight after you are done flying. They are a sequence of steps that you follow.

Susurrus, 2015-03-05:
Yes, if you land between flights that would be true. I guess I hadn't 
been seeing it that way. When I'm out with my boat I don't come back and 
dock and then replan and go again. I'm stopping the boat while at sea, 
replanning, and then starting another test. So I don't quite have the 
same workflow. I'm 1 then back and forth between 2 & 3 for X amount of 
times, and I'm usually doing 4 off and on as well to monitor things. So 
from this perspective I really see 2, 3, & 4 as flying. While you start 
at the beginning doing 1, then you do 2 & 3 for a while and maybe a 
little 4. And 4 isn't don't offline, so I don't see how this is done 
after flying.

DonLakeFlyer, 2015-03-05:
You won't always use that workflow. But for first time users it will be common. After that they'll figure out they can just do whatever they want. Closing since I think the battery is going to stay as is. And for now we'll leave the wording.

closed 2015-03-05 by DonLakeFlyer