mavlink/qgroundcontrol #1314: No audio output from QGC

Susurrus, 2015-03-04:
I'm unable to get either beeps or audio output of any kind from QGC. I've compiled in text-to-speech support as well as the annoying audio beeps whenever anything goes wrong. Sound works on my system (including using espeak directly from the command line) and there's no audio at all. I'm on Linux Fedora 20 with XFCE as my desktop running Qt 5.4.0.

I posted on the Google group, but received no help. Considering sound works otherwise on my system I'm leave to believe this is a bug in QGC and/or Qt.

Susurrus, 2015-03-06:
I'd just like to expand on this that I'm having the same problem with Windows 8.1, I get no audio output from the text-to-speech reader. Does this work for versions that other people are compiling?

dogmaphobic, 2015-03-06:
I get no audio from Windows either but I always assumed it was because it is running on a VM (and the VM sound is generally bad). Ubuntu does work, though it sounds like a Klingon with a bad case of diarrhea, which I attributed to the same VM issues above.

Susurrus, 2015-03-06:
@LorenzMeier @DonLakeFlyer Do you guys have audio output working?

thomasgubler, 2015-03-06:
I can confirm that audio works on archlinux QT 5.4.1 and espeak 1.48.04

Susurrus, 2015-03-06:
Huh. I'm debugging this on my computer by setting a breakpoint both in `GAudioOutput.cc::say()` and in `QGCAudioWorker.cpp::say()`. I've confirmed that the `GAudioOutput.cc::say()` function is called and should emit its signal that then calls the `QGCAudioWorker.cpp::say()` function in the worker thread, but that second breakpoint is never hit. So I don't think this is an espeak problem but a weird threading issue actually.

Susurrus, 2015-03-06:
And now I've seen it hit the correct breakpoint but not say anything on my Linux box. Odd.

Susurrus, 2015-03-06:
Well, on Windows 8.1 with the latest at least the talking works fine. I had it muted before and didn't realize it. So it may be a configuration issue with my Linux box. @thomasgubler Did you have to do anything beyond just install the libs and compile Qt correctly?

thomasgubler, 2015-03-07:
@Susurrus nothing special was needed

Susurrus, 2015-03-07:
I'll try and debug this on my main laptop also running Fedora. It might be something to do with how they handle speech synthesis as they use the speech-dispatcher program. I'm going to leave this open for now as I'd like to update the Fedora-specific docs once this is resolved.

fboris, 2015-08-25:
I also met this situation. I tested on my Lubuntu 14.04 laptop with Qt5.4. The difference is that espeak will say "link lost to system 1" when I plug the serial link to my laptop. However, it will only say in that time and never speak again. I confirmed it by some breakpoints.

LorenzMeier, 2015-10-25:
Audio got fixed in the last days.

closed 2015-10-25 by LorenzMeier