mavlink/qgroundcontrol #1317: QGC refuses to completely quit

Susurrus, 2015-03-04:
On 64-bit Linux, Fedora 21 (actually 20 as well), qgroundcontrol never quits when I tell it to. The UI closes, but when running it from the command line, the process never dies. It looks like there are still 7 processes running at this point but taking 0% CPU and holding 10% of my memory. I imagine there's a deadlock going on somewhere between threads? I don't know how the threading model for QGC works.

So to completely close QGC I have to force-kill it.

DonLakeFlyer, 2015-03-16:
Same on Ubuntu.

kd0aij, 2015-03-16:
I see only one process left running on Ubuntu

tail of console output:
Link deleted:  "ttyACM0"
SerialLinkLog: Already disconnected
UASLog: removeLink: "3f83b90"
UASLog: link count: 1
Deleting UAS object:  "MAV 001"
^C

DonLakeFlyer, 2015-03-21:
@dogmaphobic Gus, do you think you could look at this one. I'm not much of a Linux person. We may need to back port any fix to Stable V2.3.

dogmaphobic, 2015-03-21:
I'll take a look...

dogmaphobic, 2015-03-21:
I could not reproduce the problem. It exits just fine. I started a session from the command line, made a connection, messed with it for a bit and exited. It exited promptly:

```
Default compId:  50
Deleting UAS object:  "MAV 001"
After app delete
ubuntu:~/$
```

DonLakeFlyer, 2015-03-30:
This is being caused by audio output. It's hung up in a call to speak and the thread won't exit. I can repro. But I also can't get speak to say anything to me. Is espeak audio output working for folks?

kd0aij, 2015-03-30:
espeak works for me on 14.10, doesn't hang
Sound works in QGC also, and the voice sounds much nicer. Which one is it?

Susurrus, 2015-03-30:
espeak doesn't work for me on Fedora 21, but It works on Windows 8.1 for me.

DonLakeFlyer, 2015-03-31:
> espeak works for me on 14.10, doesn't hang

@kd0aij Can you look at a stack trace for when shutdown hangs? If it's in the GAudioOutput destructor on one of the threads than that is why it's hanging.

kd0aij, 2015-03-31:
#0  pthread_cond_wait@@GLIBC_2.3.2 () at
../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007fc86ed536fb in QWaitCondition::wait(QMutex*, unsigned long) ()
from /opt/qt54/lib/libQt5Core.so.5
#2  0x00007fc86ed522ce in QThread::wait(unsigned long) () from
/opt/qt54/lib/libQt5Core.so.5
#3  0x00000000006212a0 in GAudioOutput::~GAudioOutput() ()
#4  0x0000000000621367 in GAudioOutput::_deleteSingleton() ()
#5  0x000000000055f8fc in QGCApplication::~QGCApplication() ()
#6  0x000000000055f929 in QGCApplication::~QGCApplication() ()
#7  0x00000000004564ef in main ()

DonLakeFlyer, 2015-03-31:
Yup. That's it. My pull should fix it.

kd0aij, 2015-03-31:
nothing left hanging:

SerialLinkLog: Already disconnected
Deleting UAS object:  "MAV 001"
UASParameterCommsMgrLog: UASParameterCommsMgr destructor:  "0x26f6b38"
QThread: Destroyed while thread is still running
After app delete
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

Segmentation fault (core dumped)

DonLakeFlyer, 2015-03-31:
I don’t think the hack is going to work, bummer. It makes Qt unhappy which is then crashing out.

kd0aij, 2015-05-23:
I'm back to having to manually kill QGC on Ubuntu:
```
(gdb) bt
#0  pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007f5279a136fb in QWaitCondition::wait(QMutex*, unsigned long) ()
   from /opt/qt54/lib/libQt5Core.so.5
#2  0x00007f5279a122ce in QThread::wait(unsigned long) ()
   from /opt/qt54/lib/libQt5Core.so.5
#3  0x0000000000541650 in GAudioOutput::~GAudioOutput() ()
#4  0x00000000005416b9 in GAudioOutput::~GAudioOutput() ()
#5  0x0000000000549aec in QGCApplication::~QGCApplication() ()
#6  0x0000000000549b49 in QGCApplication::~QGCApplication() ()
#7  0x0000000000440641 in main ()
```

DonLakeFlyer, 2015-05-25:
This was never fixed. Someone with Linux experience is going to have to try to fix it. espeak is hanging preventing QGC from shutting down. I can't get espeak working at all on my machine.

closed 2015-10-25 by DonLakeFlyer