mavlink/qgroundcontrol #1326: [new toolbar] ARMED and EMERGENCY are easy to confuse

Susurrus, 2015-03-06:
Since the colors of ARMED and EMERGENCY are both the same color of red, they're easy to confuse. I would expect red to denote an error condition, not necessarily one that is just a warning. Maybe yellow can be used instead for ARMED, like it had been for the old toolbar?

dogmaphobic, 2015-03-06:
Certainly. The colors in the tool bar are generally experimental as they are not part of the main QGC palette yet (other then buttons and window). When I first started, I had it orange. We need to figure out these colors and make it part of the main QGC palette. @DonLakeFlyer Can you comment on these? Should they be part of QGCPalette?

DonLakeFlyer, 2015-03-06:
I would say no, these shouldn't be in the palette. Reason: They should stay the same color in both the light and dark themes. That's kind of the rule for now. Kind of the same as the little red/green indicator dots that are all over setup. Those aren't in the palette either. A reason to break that rule would be if you have a useful color that you want to keep same across multiple Qml files, even if it doesn't change in light/dark.

closed 2015-03-08 by LorenzMeier