mavlink/qgroundcontrol #1328: parameters view distorted

thomasgubler, 2015-03-06:
[image]

The bar on the left moves towards the center and the parameter widget is tiny and strangely placed.

Here is the Summary screen for comparison.
[image]

This is with QT 5.4.1,

thomasgubler, 2015-03-06:
Same for flight modes:
[image]

LorenzMeier, 2015-03-06:
I have the same effect on a laptop screen.

dogmaphobic, 2015-03-06:
I just assumed it to be WIP...

DonLakeFlyer, 2015-03-06:
Yup. Non-qml setup screens are a bit out of whack right now with respect to positioning. I had been leaving them only since I was converting them over one at a time. With Flight Modes to Qml now it's just parameters left. So it's easy enough to fix just that one. Conversion of that screen to Qml is not high pro.

DonLakeFlyer, 2015-03-06:
Fixed. New Flight Mode fixes Flight Mode.

closed 2015-03-06 by DonLakeFlyer