mavlink/qgroundcontrol #1338: Radio config for SiK radios

LorenzMeier, 2015-03-08:
I'm starting to collect notes here. I don't think I'll have the bandwidth to do the full UI, but I will bring up the required Travis setup to provide binaries and I'm happy to help with the bootloader. The PX4 boot loaders and SiK bootloader are essentially the same implementation, so we should be able to get something together reasonably quick.

The configuration is even easier: Its just AT commands:
http://www.pixhawk.org/peripherals/radio-modems/start#dr_usb_radio

yekailove, 2015-03-25:
Woo~How can I useing the 3DR Radio 915MHz communicate with QGroundControl?

yekailove, 2015-03-25:
[image]
[image]

DonLakeFlyer, 2015-03-26:
Nothing special with QGC and the 3DR Radio. I’d go to the 3dr site and read up on how you use it with either mission planner or APM Planner. It’s pretty much the same in QGC. You’ll need to install the windows driver if you are using windows.

nopeppermint, 2015-03-31:
@yekailove three sites about the 3DR Radio configuration:
- http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-version-2/
- http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-advanced-configuration-and-technical-information/
- http://planner.ardupilot.com/wiki/other-project-and-common-topics/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-v1/

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier