mavlink/qgroundcontrol #1345: Run instructions for Linux

LorenzMeier, 2015-03-10:
It seems to run the pre-built binary, one has to install:

```
sudo apt-get install libstdc++6
```

LorenzMeier, 2015-03-10:
I've added instructions next to the download link. If that's good enough feel free to close.
http://qgroundcontrol.org/downloads

NatalieUW2, 2015-03-10:
I'm running 14.04 Ubuntu and am having issues getting the software to run. I re-installed libstdc++6 (along with libsdl1.2debian and libqt5serialport5), but am still getting the following error when I try to run qgroundcontrol:

    ./qgroundcontrol: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version 'GLIBCXX_3.4.20'
    not found (required by ./qgroundcontrol)

It appears that I have GLIBCXX_3.4.1 through 3.4.19 (based on the output of:
'strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX'), but not 3.4.20. Any suggestions on how to get this working?

DonLakeFlyer, 2015-03-17:
It comes with the latest version of gcc.

DonLakeFlyer, 2015-03-20:
I changed the downloads page to references the readme and updated the readme to include this.

closed 2015-03-20 by DonLakeFlyer