mavlink/qgroundcontrol #1366: Can't add most message values to the Info View's Quick tab

Susurrus, 2015-03-17:
This used to be possible, I think maybe adding the Fact system broke this? @DonLakeFlyer Can you confirm this?

I'd really like to show for example: crosstrack error, roll, and yaw and none of these are options. I know it *should* be possible.

DonLakeFlyer, 2015-03-17:
No changes there that I know of.

Susurrus, 2015-03-17:
I used to get a huge list of values, now I only see 2 different messages. Does this work for you?

DonLakeFlyer, 2015-03-17:
Nope. List should come from mavlinkdecoder.

dogmaphobic, 2015-03-17:
Is this what you're talking about?

[image]

DonLakeFlyer, 2015-03-17:
Yup. It should have values in there for each thing that mavlnkdecoder decodes and signals a value change on.

Susurrus, 2015-03-19:
Exactly. I only see the HEARTBEAT and SYS_STATUS messages. I know this used to work.

DonLakeFlyer, 2015-03-19:
Don't have GitHub right now. Can you mark this for stable v2.3? So it gets fixed before next stable

dogmaphobic, 2015-03-19:
It's all there now (#1377)

DonLakeFlyer, 2015-03-19:
Fixed

closed 2015-03-19 by DonLakeFlyer