mavlink/qgroundcontrol #1371: Problem with making RN-XV work with QGC

nimaaghli, 2015-03-18:
Hi,
I have configured RN-XV  with pixhawk the boud rate is 57600 ,
as soon as im connecting to port in udp i see the UAV in my list so i think im, getting MAV messages correct but after 2 second the program freezes and when i check the terminal im getting this message repeating
 UDP:Adding HOST 169.254.1.25:2000"
169.254.1.25 is the ip adress of RN-XV
Port 2000 is the port of RN-XV.

DonLakeFlyer, 2015-03-18:
Exact sequence of steps for me to reproduce this please, plus what OS? I'm not familiar with UDP so you will need to walk me through all of it.

dogmaphobic, 2015-03-18:
I'm adding UDP to my "slave" MavLink. I will be able to test this in the next couple of days...

nimaaghli, 2015-03-19:
Problem solved buy change UDPlink.cc and UDPlink.h

nimaaghli, 2015-03-19:
in void UDPLink::readBytes() inside the while its trying to add _config->addHost(host); each time and it is causing program to stop working i changed the code this way 
if (!_hosts.contains(sender.toString())){
            QString host(sender.toString() + ":" + QString("%1").arg((int)senderPort));
            _config->addHost(host);
            _hosts[sender.toString()] = senderPort;
        }
so it only adds the host to list once

dogmaphobic, 2015-03-19:
The fix in in PR #1374. Thanks for spotting it (and giving me the incentive to finish my own side project).

DonLakeFlyer, 2015-03-20:
@nimaaghli Could you double-check the latest releases or master build to make sure this is fixed for you now?

nimaaghli, 2015-03-20:
@DonLakeFlyer Yes i bulid the last releases and tested with single pixhawk it works great . Will test on multiple pixhawk tomorrow.

DonLakeFlyer, 2015-03-20:
Thanks @nimaaghli. I'm closing this now and you can re-open if you still find a problem.

closed 2015-03-20 by DonLakeFlyer