mavlink/qgroundcontrol #1384: Red messges for non critical messges

nimaaghli, 2015-03-22:
Some Messages are not critical but they are in red color i have got this as red color 
takeoff to 10.00 meters above home

dogmaphobic, 2015-03-23:
The red icon color reflects the presence of *any* error while the count is the total count of new messages. That is, if you only have normal messages, the icon remains blue. As new messages arrive, the counter goes up. Once a *warning* message arrives, the icon turns orange, and once an *error* message arrives, the icon turns red. Even if subsequent messages arriving are normal, the icon remains red to alert you that there is an error message in there. Once you click on the icon and the message list is displayed, it all resets back to normal.

With that said, are you seeing the icon red when there is no error messages? This would be an error, and I can't quite see how that could happen. If you see an error message that should not be an error, that's a different problem and it needs to be handle in the message origin. Which one is it?

nimaaghli, 2015-03-24:
yes you are right please check the issue#1385 the discussion with @DonLakeFlyer was that if the msg is error or no !

closed 2017-08-07 by nimaaghli