mavlink/qgroundcontrol #1386: GUI changes while changing MAV in multi drone

nimaaghli, 2015-03-24:
Hello, 
While having multi drone and clicking on them to control the GUI has some changes on menu the arrow shapes dissapear and show up again when mouse goes over them

nimaaghli, 2015-03-24:
[image]
[image]

DonLakeFlyer, 2015-03-26:
@dogmaphobic Something for you Gus

DonLakeFlyer, 2015-03-26:
@dogmaphobic I was just talking about the toolbar button redraw problems for you to look at. Can you pull that out into a separate issue?

dogmaphobic, 2015-03-26:
Duh... This entire time I was assuming it was an issue within the map view. It didn't even occur to me to look at the tool bar and see the chevron buttons are not there. It's the OpenGL Canvas issue (QtQuick inside a QWidget). I know how to get around it, I just need to be able to reproduce it. I will wait for your #1392 to get merged and I will take a look.

closed 2015-10-25 by DonLakeFlyer