mavlink/qgroundcontrol #1390: Altitude and vZ displayed on Primary Flight Display even if invalid

Susurrus, 2015-03-26:
For the PFD, generally things aren't displayed or are at least handled nicely if there is invalid data for them. As I'm using a surface vessel, I odn't care about altitude or climb rate, so I output NaNs for these values. It still displays the altitude meter and the vZ indicator. Most annoying, the vZ indicator flickers displaying briefly, then disappearing briefly, only to come back. So at least the vZ, an invalid value isn't handled nicely, where at least with the Altitude it just displays 0.

I propose that the Altitude display -- or not be displayed if ti doesn't exist. vZ should also not be displayed if Altitude is NaN, otherwise if it isn't, then it shouldn't be displayed if it's a NaN.

dogmaphobic, 2015-04-08:
I'm assuming this is no longer an issue with the new PFD?

Susurrus, 2015-04-08:
I haven't had a chance to test. I'm currently doing extensive data collection with my robot and can't risk things going wrong with QGC given its rapid pace of development. If no one else is willing to follow up on this, then it should just be closed.

closed 2015-12-03 by DonLakeFlyer