mavlink/qgroundcontrol #1399: Crash in DebugConsole

dogmaphobic, 2015-03-28:
Putting it down here before I forget.

To reproduce:

Start QGC and then plug in PX4 (USB). Click connect while the PX4 is booting. Nothing will happen as QGC is waiting for the bootloader to go away. Hit disconnect and then connect again. DebugConsole crashes in ```DebugConsole::updateTrafficMeasurements()``` as it tries to use a link pointer that is no longer valid.

That whole DebugConsole may need some rethinking as the UI is all broken and its functionality is dubious at best.

LorenzMeier, 2015-03-30:
We need a text terminal for the system with basic VT100 capabilities. It's the main interface during development. The current implementation is indeed messed up, but it would be nice to get a proper console setup going.

dogmaphobic, 2015-03-30:
I’m not dismissing the need, rather, the implementation. Not to mention the redundancy with the Terminal widget. It’s on my todo list...

g

DonLakeFlyer, 2015-03-30:
We need one terminal to rule them all!

DonLakeFlyer, 2015-04-05:
This is fixed now

closed 2015-04-05 by DonLakeFlyer