mavlink/qgroundcontrol #1402: QGC segfaults on disconnect: Ubuntu 14.10

kd0aij, 2015-03-30:
latest master
```
SerialLinkLog: Closing Port # 194 "ttyACM0"
MAVLinkProtocolLog: _linkStatusChanged "44a2030" false
UASLog: _linkDisconnected: "/dev/ttyACM0"
UASLog: link count: 1
SerialLinkLog: Already disconnected
Deleting UAS object:  "MAV 001"
Segmentation fault (core dumped)

#0  0x00007f3754347370 in ?? () from /opt/qt54/qml/QtQuick/Controls/libqtquickcontrolsplugin.so
#1  0x00007f378b5ba8eb in QObject::disconnect(QObject const*, char const*, QObject const*, char const*) () from /opt/qt54/lib/libQt5Core.so.5
#2  0x000000000060d50f in DebugConsole::linkSelected(int) ()
#3  0x00000000007a2dcb in DebugConsole::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#4  0x00007f378b5b81a1 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /opt/qt54/lib/libQt5Core.so.5
#5  0x00007f378cba4951 in QComboBox::currentIndexChanged(int) ()
   from /opt/qt54/lib/libQt5Widgets.so.5
#6  0x00007f378cba6b61 in ?? () from /opt/qt54/lib/libQt5Widgets.so.5
#7  0x00007f378cbab601 in ?? () from /opt/qt54/lib/libQt5Widgets.so.5
#8  0x00007f378b5b81a1 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /opt/qt54/lib/libQt5Core.so.5
#9  0x00007f378b6334e4 in QAbstractItemModel::rowsRemoved(QModelIndex const&, int, int, QAbstractItemModel::QPrivateSignal) () from /opt/qt54/lib/libQt5Core.so.5
#10 0x00007f378b536592 in QAbstractItemModel::endRemoveRows() ()
   from /opt/qt54/lib/libQt5Core.so.5
#11 0x00007f378c31caf2 in QStandardItem::removeRows(int, int) ()
   from /opt/qt54/lib/libQt5Gui.so.5
#12 0x00007f378c31cbd4 in QStandardItemModel::removeRows(int, int, QModelIndex const&) ()
   from /opt/qt54/lib/libQt5Gui.so.5
#13 0x000000000060d468 in DebugConsole::removeLink(LinkInterface*) ()
---Type <return> to continue, or q <return> to quit---
#14 0x00007f378b5b7c1f in QMetaObject::activate(QObject*, int, int, void**) ()
   from /opt/qt54/lib/libQt5Core.so.5
#15 0x0000000000795482 in LinkManager::linkDisconnected(LinkInterface*) ()
#16 0x00007f378b5b8fa6 in QObject::event(QEvent*) () from /opt/qt54/lib/libQt5Core.so.5
#17 0x00007f378ca9b86c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /opt/qt54/lib/libQt5Widgets.so.5
#18 0x00007f378caa0736 in QApplication::notify(QObject*, QEvent*) ()
   from /opt/qt54/lib/libQt5Widgets.so.5
#19 0x00007f378b588945 in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /opt/qt54/lib/libQt5Core.so.5
#20 0x00007f378b58a7df in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
    () from /opt/qt54/lib/libQt5Core.so.5
#21 0x00007f378b5e0853 in ?? () from /opt/qt54/lib/libQt5Core.so.5
#22 0x00007f3788259c5d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007f3788259f48 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007f3788259ffc in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007f378b5dfccc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /opt/qt54/lib/libQt5Core.so.5
#26 0x00007f378b5868ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /opt/qt54/lib/libQt5Core.so.5
#27 0x00007f378cc851bd in QDialog::exec() () from /opt/qt54/lib/libQt5Widgets.so.5
#28 0x00000000007447ce in QGCFileDialog::getSaveFileName(QWidget*, QString const&, QString const&---Type <return> to continue, or q <return> to quit---
, QString const&, QString const&, bool, QFlags<QFileDialog::Option>) ()
#29 0x00000000005614be in QGCApplication::saveTempFlightDataLogOnMainThread(QString) ()
#30 0x00000000005911af in QtPrivate::QSlotObject<void (QGCApplication::*)(QString), QtPrivate::List<QString, void>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) ()
#31 0x00007f378b5b7c1f in QMetaObject::activate(QObject*, int, int, void**) ()
   from /opt/qt54/lib/libQt5Core.so.5
#32 0x0000000000797545 in MAVLinkProtocol::saveTempFlightDataLog(QString) ()
#33 0x000000000058c680 in MAVLinkProtocol::_stopLogging() ()
#34 0x000000000058d994 in MAVLinkProtocol::_linkStatusChanged(LinkInterface*, bool) ()
#35 0x00007f378b5b8fa6 in QObject::event(QEvent*) () from /opt/qt54/lib/libQt5Core.so.5
#36 0x00007f378ca9b86c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /opt/qt54/lib/libQt5Widgets.so.5
#37 0x00007f378caa0736 in QApplication::notify(QObject*, QEvent*) ()
   from /opt/qt54/lib/libQt5Widgets.so.5
#38 0x00007f378b588945 in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /opt/qt54/lib/libQt5Core.so.5
#39 0x00007f378b58a7df in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
    () from /opt/qt54/lib/libQt5Core.so.5
#40 0x00007f378b5e0853 in ?? () from /opt/qt54/lib/libQt5Core.so.5
#41 0x00007f3788259c5d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#42 0x00007f3788259f48 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#43 0x00007f3788259ffc in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
---Type <return> to continue, or q <return> to quit---
#44 0x00007f378b5dfccc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /opt/qt54/lib/libQt5Core.so.5
#45 0x00007f378b5868ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /opt/qt54/lib/libQt5Core.so.5
#46 0x00007f378b58df16 in QCoreApplication::exec() () from /opt/qt54/lib/libQt5Core.so.5
#47 0x00000000004564ca in main ()
```

DonLakeFlyer, 2015-03-30:
Thanks a ton for the stack trace Mark. I’m pretty sure I know what is going on here.

kd0aij, 2015-03-30:
I'm glad you know; I don't have a clue :)

DonLakeFlyer, 2015-03-30:
@kd0aij Can you check this again?

kd0aij, 2015-03-30:
pulled upstream:
commit f053edc8dff48b1784d1a7a0e7cbe52fc39a2258
Merge: 8dd7b72 f712a7f
Author: Don Gagne <don@thegagnes.com>
Date:   Mon Mar 30 10:44:17 2015 -0700

    Merge pull request #1403 from DonLakeFlyer/DebugConsole
    
    Track links using shared pointed

Still segfaults on disconnect:
```
SerialLinkLog: Closing Port # 192 "ttyACM0"
MAVLinkProtocolLog: _linkStatusChanged "3fcb720" false
UASLog: _linkDisconnected: "/dev/ttyACM0"
UASLog: link count: 1
Deleting UAS object:  "MAV 001"
SerialLinkLog: Already disconnected
Segmentation fault (core dumped)

#0  0x0000000003fcb618 in ?? ()
#1  0x00007f3ad466a8eb in QObject::disconnect(QObject const*, char const*, QObject const*, char const*) ()
   from /opt/qt54/lib/libQt5Core.so.5
#2  0x000000000060d42f in DebugConsole::linkSelected(int) ()
#3  0x00000000007a533b in DebugConsole::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#4  0x00007f3ad46681a1 in QMetaObject::activate(QObject*, int, int, void**) () from /opt/qt54/lib/libQt5Core.so.5
#5  0x00007f3ad5c54951 in QComboBox::currentIndexChanged(int) () from /opt/qt54/lib/libQt5Widgets.so.5
#6  0x00007f3ad5c56b61 in ?? () from /opt/qt54/lib/libQt5Widgets.so.5
#7  0x00007f3ad5c5b601 in ?? () from /opt/qt54/lib/libQt5Widgets.so.5
#8  0x00007f3ad46681a1 in QMetaObject::activate(QObject*, int, int, void**) () from /opt/qt54/lib/libQt5Core.so.5
#9  0x00007f3ad46e34e4 in QAbstractItemModel::rowsRemoved(QModelIndex const&, int, int, QAbstractItemModel::QPrivateSignal) () from /opt/qt54/lib/libQt5Core.so.5
#10 0x00007f3ad45e6592 in QAbstractItemModel::endRemoveRows() () from /opt/qt54/lib/libQt5Core.so.5
#11 0x00007f3ad53ccaf2 in QStandardItem::removeRows(int, int) () from /opt/qt54/lib/libQt5Gui.so.5
#12 0x00007f3ad53ccbd4 in QStandardItemModel::removeRows(int, int, QModelIndex const&) ()
   from /opt/qt54/lib/libQt5Gui.so.5
#13 0x000000000060d2b8 in DebugConsole::removeLink(LinkInterface*) ()
#14 0x00007f3ad4667c1f in QMetaObject::activate(QObject*, int, int, void**) () from /opt/qt54/lib/libQt5Core.so.5
#15 0x00000000007979f2 in LinkManager::linkDisconnected(LinkInterface*) ()
#16 0x00007f3ad4668fa6 in QObject::event(QEvent*) () from /opt/qt54/lib/libQt5Core.so.5
#17 0x00007f3ad5b4b86c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /opt/qt54/lib/libQt5Widgets.so.5
#18 0x00007f3ad5b50736 in QApplication::notify(QObject*, QEvent*) () from /opt/qt54/lib/libQt5Widgets.so.5
#19 0x00007f3ad4638945 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /opt/qt54/lib/libQt5Core.so.5
#20 0x00007f3ad463a7df in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
   from /opt/qt54/lib/libQt5Core.so.5
#21 0x00007f3ad4690853 in ?? () from /opt/qt54/lib/libQt5Core.so.5
#22 0x00007f3ad1309c5d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007f3ad1309f48 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007f3ad1309ffc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007f3ad468fccc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /opt/qt54/lib/libQt5Core.so.5
#26 0x00007f3ad46368ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /opt/qt54/lib/libQt5Core.so.5
#27 0x00007f3ad5d351bd in QDialog::exec() () from /opt/qt54/lib/libQt5Widgets.so.5
#28 0x00000000007447ae in QGCFileDialog::getSaveFileName(QWidget*, QString const&, QString const&, QString const&, QString const&, bool, QFlags<QFileDialog::Option>) ()
#29 0x000000000056133e in QGCApplication::saveTempFlightDataLogOnMainThread(QString) ()
#30 0x000000000059102f in QtPrivate::QSlotObject<void (QGCApplication::*)(QString), QtPrivate::List<QString, void>, void>::---Type <return> to continue, or q <return> to quit---
impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) ()
#31 0x00007f3ad4667c1f in QMetaObject::activate(QObject*, int, int, void**) () from /opt/qt54/lib/libQt5Core.so.5
#32 0x0000000000799ab5 in MAVLinkProtocol::saveTempFlightDataLog(QString) ()
#33 0x000000000058c4c0 in MAVLinkProtocol::_stopLogging() ()
#34 0x000000000058d7d4 in MAVLinkProtocol::_linkStatusChanged(LinkInterface*, bool) ()
#35 0x00007f3ad4668fa6 in QObject::event(QEvent*) () from /opt/qt54/lib/libQt5Core.so.5
#36 0x00007f3ad5b4b86c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /opt/qt54/lib/libQt5Widgets.so.5
#37 0x00007f3ad5b50736 in QApplication::notify(QObject*, QEvent*) () from /opt/qt54/lib/libQt5Widgets.so.5
#38 0x00007f3ad4638945 in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /opt/qt54/lib/libQt5Core.so.5
#39 0x00007f3ad463a7df in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
   from /opt/qt54/lib/libQt5Core.so.5
#40 0x00007f3ad4690853 in ?? () from /opt/qt54/lib/libQt5Core.so.5
#41 0x00007f3ad1309c5d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#42 0x00007f3ad1309f48 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#43 0x00007f3ad1309ffc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#44 0x00007f3ad468fccc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /opt/qt54/lib/libQt5Core.so.5
#45 0x00007f3ad46368ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /opt/qt54/lib/libQt5Core.so.5
#46 0x00007f3ad463df16 in QCoreApplication::exec() () from /opt/qt54/lib/libQt5Core.so.5
#47 0x00000000004564da in main ()
```

DonLakeFlyer, 2015-03-30:
> Author: Don Gagne don@thegagnes.com <mailto:don@thegagnes.com>

Are you manually disconnecting or is the system disconnecting when you close QGC?

kd0aij, 2015-03-30:
Pressing the "Disconnect" button.
But: After disconnecting/reconnecting Pixhawk on USB, can't reproduce the
segfault.

Still have to ^C the qgc process after exiting, though.

DonLakeFlyer, 2015-03-30:
I think it might be fixed for real now. Also I cannot repro the shutdown problems you are having. Tried a number of times as has Gus.

kd0aij, 2015-03-30:
Could it have something to do with sound? If I "mute all audio output" on the File:settings menu, QGC exits cleanly.

DonLakeFlyer, 2015-03-30:
No. It’s just like of the draw with the memory manager so it’s going to move around when you do other things with the system.

DonLakeFlyer, 2015-03-30:
> Could it have something to do with sound? If I "mute all audio output" on the File:settings menu, QGC exits cleanly.

Sorry my previous post was reading the wrong thing. Let me see if audio is the problem. Could me.

dogmaphobic, 2015-03-30:
> Could it have something to do with sound?

Interesting... I have disabled audio on Ubuntu altogether because it would spend a whole lot of time displaying thousands of error messages on the console related to audio. I don't remember the actual error message. I just assumed my audio setup was messed up and just disabled it.

DonLakeFlyer, 2015-03-30:
Yup its audio. I can repro now.

kd0aij, 2015-07-20:
What's the status of this issue? I've had to disable audio on Linux also, because QGC never shuts down if it is enabled. (At one time it was segfaulting, which at least didn't leave a process hanging, but that workaround has apparently been removed)
Which also means I can't contribute to testing whether audio is working properly.

DonLakeFlyer, 2015-07-20:
I tried fixing it, but it’s going to need someone familiar with the linux speech stuff to figure it out.

LorenzMeier, 2015-08-01:
@kd0aij Audio shutdown should work now. Could you try again?

closed 2015-10-25 by DonLakeFlyer