mavlink/qgroundcontrol #1407: Using Arduplane and APM with CGS: home-location and waypont 0 don't work

severinleuenberger, 2015-03-31:
I have historically some drones with the older APM2.6 and 3.1 Hardware with arduplane on it (no PX4). So I like to connect them to QGC as well. Basically it works! 

But 2 issues are a problem: 
1.       when I try to define the Home-Location, it doesn’t work.
2.       the waypoint number is an issue. With QGC it starts with 0 and on mission planner with arduplane the waypoints start with 1. What happens, is: the waypoint 0 is not recognized by arduplane.

Please implement a fix with some logic like “if APM then waypoint start count 1 else 0” 
and the Home-Location Fix would be great as well. 
 
Best whishes, 
Severin

LorenzMeier, 2015-03-31:
We will instead fix APM's non-standard use and only support the new WP protocol at the DroneCode level.

closed 2015-03-31 by LorenzMeier

severinleuenberger, 2015-03-31:
sounds good, where do i have to look for the developement? Mavlink-protocol?