mavlink/qgroundcontrol #1417: map cache doesnt work on v2.3.0 stable

severinleuenberger, 2015-04-04:
I use Windows 8 on a 64bit version with v2.3.0 (stable release) of QGroundControl
Downloaded 2 weeks ago. 

With internet-connection - everything works fine, but even when i have cached google hybrid until zoom level 20, the map view doesn show any map, GPS health is good (8 satelites). 

Connected with 433 3rd link to a dropix(pixhawk) with stable release of PX4. 

Can you please fix this?

DonLakeFlyer, 2015-04-05:
@dogmaphobic I know we're moving away from this code but that will take a while. And you were silly enough to probably know more than anyone else about the old map cache code now. So could you take a look at this when you have time.

dogmaphobic, 2015-04-05:
The irony is that I could not get *Hybrid* map to work yet with the new code. I'll take a look, though my only Windows "machine" is a VM running Windows 7.

dogmaphobic, 2015-04-08:
Hmm... I can't reproduce this. Using google hybrid map, I've selected a rectangle around my house and let it cache up to zoom level 18. I shut QGC down, disabled outgoing network connections and booted it back up. I also added some debugging output to show what's going on. When I hit *Go Home*, it all comes up fine from the cache:
```
Cached Image:  73749 x 103092 Z: 18
Cached Image:  73749 x 103094 Z: 18
Cached Image:  73749 x 103095 Z: 18
Cached Image:  73749 x 103092 Z: 18
Cached Image:  73749 x 103096 Z: 18
Cached Image:  73749 x 103093 Z: 18
Cached Image:  73749 x 103094 Z: 18
Cached Image:  73749 x 103095 Z: 18
Cached Image:  73749 x 103096 Z: 18
Cached Image:  73749 x 103093 Z: 18
```
As I zoom in, it starts to barf (as it should) as I only cached up to zoom level 18:
```
No Cached Image found for:  147491 x 206189 Z: 19
No Cached Image found for:  147491 x 206187 Z: 19
No Cached Image found for:  147491 x 206188 Z: 19
No Cached Image found for:  147491 x 206186 Z: 19
No Cached Image found for:  147491 x 206190 Z: 19
```

dogmaphobic, 2015-04-08:
@severinleuenberger Could you please delete your cache file, cache a new small rectangle and send me the file so I can take a look at? The file is in your home directory in a subdirectory called mapscache as in:
```
/Users/your_name/mapscache/Data.qmdb
```
(using the the appropriate root path according to your OS)

severinleuenberger, 2015-04-08:
As i deleted the cache file and rebuilt it, it seems to work now. Maybe I closed QGC completely during caching and corrupted the file. 
Thanks for the hint to delete the file

closed 2015-04-15 by DonLakeFlyer