mavlink/qgroundcontrol #1423: Ubuntu Wont Install qgroundcontrol

iestatic, 2015-04-07:
Hello I have ubuntu 14.04 and I am trying to install qgroundcontrol. I finally figured out how to extract the files but when I try to run it (even after setting the file to executable) nothing happens. Literally nothing.. HELP!

DonLakeFlyer, 2015-04-07:
Did you install the dependencies?

iestatic, 2015-04-07:
Yes. I did everything the site told me to do. I think?

DonLakeFlyer, 2015-04-07:
Which version are you installing?

Thomas: Have you unpacked any builds lately? Wondering if you are running into problems.

iestatic, 2015-04-07:
The newest one on the site. The stable one. I was originally trying to install it on my windows 7 but I couldn't get it to connect to the card for my ar drone. I tried everything I could and just decided to install Ubuntu and try. I haven't really tried to install anything else other than maybe a browser it went fine.

DonLakeFlyer, 2015-04-07:
If you can’t get it to connect on the Windows version and you are not a Linux whiz you are likely to have more problems getting it connected on Ubuntu. Linux is more finicky with permissions than windows is.

iestatic, 2015-04-07:
Any idea on why it won't connect to the device on widows?

DonLakeFlyer, 2015-04-08:
If you’re using an ar drone then I‘d try to ask on an ar drone forum. There aren’y many folks with an ar drone here to help out.

DonLakeFlyer, 2015-04-08:
I'm going to close this unless other people start reporting ubuntu is broken as well.

closed 2015-04-08 by DonLakeFlyer