mavlink/qgroundcontrol #1427: PX4 Config: Warn user / let him confirm if no external mag is available during setup

LorenzMeier, 2015-04-08:
This is a mix between situational awareness, user interaction and firmware side: On many quads the internal mag shows bad performance and shouldn't be used. The shortest path to prevent this failure scenario needs just a QGC change: If during the mag calibration only internal mags are found, warn the user with a yes / no dialog and reset the MAG ID params if the user didn't accept the change.

The text could read something like: "No external magnetometer found during calibration. If you have a GPS + magnetometer module installed (e.g. 3DR GPS), this indicates an issue with connectivity / cabling. Using only the internal mag will likely deteriorate flight performance. Click yes to use the internal mag, click no to abort and re-check your cabling."

DonLakeFlyer, 2015-04-08:
Internal mag is always and only mag0 right?

LorenzMeier, 2015-04-08:
Not necessarily. The right approach would be to probe at this point:
https://github.com/PX4/Firmware/blob/master/src/modules/commander/mag_calibration.cpp#L128

Using ```MAGIOCGEXTERNAL``` IOCTL to work out if the mag is internal or external. That said I was wrong and it needs a FW change to alter a string and a QGC change to parse that or something along these lines.

DonLakeFlyer, 2015-04-08:
You can determine if a mag is internal by the fact that the _ID param is set but the _ROT param is <0.

closed 2015-06-02 by LorenzMeier