mavlink/qgroundcontrol #1428: crash when reconnect is occured via serial link.

mihadyuk, 2015-04-08:
Qgroundcontrol crashes when user reconnects to board and presses button "get" on onboard parameters tab.

OS: Archlinux
Linux 3.19.2-1-ARCH #1 SMP PREEMPT Wed Mar 18 16:21:02 CET 2015 x86_64 GNU/Linux

Qt5 version: 5.4.1-2
Qt4 version: 4.8.6-4

I built the qground from master on github repo. We used our own boards to check the problem and the result was fail. It seems no matter which board is used.

commit sha1: c315ff4e43e8d4e3a1ab3cca105efd9ed959a51a

Steps to reproduce:
1. Connect board to PC via serial link.
2. Run qgroundcontrol from terminal and choose page "Analyze" that is on main window.
3. Press button "Connect" and then press "Get" on onboard parameters tab.
4. Press "Disconnect".
5. Press "Connect" and then press "Get" on onboard parameters tab. At this point crash is occured. "Segmentation fault" is written on terminal.

DonLakeFlyer, 2015-04-08:
Can you get a stack trace?

mihadyuk, 2015-04-09:
Threads: #1 qgroundcontrol

0	QGCBaseParamWidget::requestOnboardParamsUpdate	QGCBaseParamWidget.cc	95	0x79fba1
1	QGCBaseParamWidget::qt_static_metacall	moc_QGCBaseParamWidget.cpp	119	0x8a24d9
2	QMetaObject::activate(QObject*, int, int, void**)	/usr/lib/libQt5Core.so.5		0x7ffff531f309
3	QAbstractButton::clicked(bool)	/usr/lib/libQt5Widgets.so.5		0x7ffff69c2a62
4	??	/usr/lib/libQt5Widgets.so.5		0x7ffff6715ba5
5	??	/usr/lib/libQt5Widgets.so.5		0x7ffff67167c9
6	QAbstractButton::mouseReleaseEvent(QMouseEvent*)	/usr/lib/libQt5Widgets.so.5		0x7ffff6716984
7	QWidget::event(QEvent*)	/usr/lib/libQt5Widgets.so.5		0x7ffff665de08
8	QApplicationPrivate::notify_helper(QObject*, QEvent*)	/usr/lib/libQt5Widgets.so.5		0x7ffff661aacc
9	QApplication::notify(QObject*, QEvent*)	/usr/lib/libQt5Widgets.so.5		0x7ffff66207d1
10	QCoreApplication::notifyInternal(QObject*, QEvent*)	/usr/lib/libQt5Core.so.5		0x7ffff52eed1b
11	QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool)	/usr/lib/libQt5Widgets.so.5		0x7ffff661f2ea
12	??	/usr/lib/libQt5Widgets.so.5		0x7ffff6676530
13	??	/usr/lib/libQt5Widgets.so.5		0x7ffff66786a3
14	QApplicationPrivate::notify_helper(QObject*, QEvent*)	/usr/lib/libQt5Widgets.so.5		0x7ffff661aacc
15	QApplication::notify(QObject*, QEvent*)	/usr/lib/libQt5Widgets.so.5		0x7ffff661ff10
16	QCoreApplication::notifyInternal(QObject*, QEvent*)	/usr/lib/libQt5Core.so.5		0x7ffff52eed1b
17	QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)	/usr/lib/libQt5Gui.so.5		0x7ffff5c78752
18	QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)	/usr/lib/libQt5Gui.so.5		0x7ffff5c7a305
19	QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Gui.so.5		0x7ffff5c5f1ef
20	??	/usr/lib/qt/plugins/platforms/libqxcb.so		0x7fffea3071d0
21	g_main_context_dispatch	/usr/lib/libglib-2.0.so.0		0x7ffff281d71d
22	??	/usr/lib/libglib-2.0.so.0		0x7ffff281da08
23	g_main_context_iteration	/usr/lib/libglib-2.0.so.0		0x7ffff281dabc
24	QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff5347077
25	QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff52ec7b2
26	QCoreApplication::exec()	/usr/lib/libQt5Core.so.5		0x7ffff52f418c
27	main	main.cc	179	0x57e5d3

Threads: #2 QQmlDebugServer

0	poll	/usr/lib/libc.so.6		0x7ffff3d6d4ed
1	??	/usr/lib/libglib-2.0.so.0		0x7ffff281d9a4
2	g_main_context_iteration	/usr/lib/libglib-2.0.so.0		0x7ffff281dabc
3	QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff534708c
4	QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff52ec7b2
5	QThread::exec()	/usr/lib/libQt5Core.so.5		0x7ffff50fe5d4
6	??	/usr/lib/libQt5Qml.so.5		0x7ffff5a539a8
7	??	/usr/lib/libQt5Core.so.5		0x7ffff510355e
8	start_thread	/usr/lib/libpthread.so.0		0x7ffff4864374
9	clone	/usr/lib/libc.so.6		0x7ffff3d7627d

Threads: #3 QQmlThread
0	poll	/usr/lib/libc.so.6		0x7ffff3d6d4ed
1	??	/usr/lib/libglib-2.0.so.0		0x7ffff281d9a4
2	g_main_context_iteration	/usr/lib/libglib-2.0.so.0		0x7ffff281dabc
3	QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff5347077
4	QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff52ec7b2
5	QThread::exec()	/usr/lib/libQt5Core.so.5		0x7ffff50fe5d4
6	??	/usr/lib/libQt5Qml.so.5		0x7ffff5a34835
7	??	/usr/lib/libQt5Core.so.5		0x7ffff510355e
8	start_thread	/usr/lib/libpthread.so.0		0x7ffff4864374
9	clone	/usr/lib/libc.so.6		0x7ffff3d7627d

Threads: #4 SerialLink

0	select	/usr/lib/libc.so.6		0x7ffff3d6f1b3
1	??	/usr/lib/libQt5SerialPort.so.5		0x7ffff55aee3b
2	??	/usr/lib/libQt5SerialPort.so.5		0x7ffff55b08c1
3	SerialLink::run	SerialLink.cc	150	0x5bcfc0
4	??	/usr/lib/libQt5Core.so.5		0x7ffff510355e
5	start_thread	/usr/lib/libpthread.so.0		0x7ffff4864374
6	clone	/usr/lib/libc.so.6		0x7ffff3d7627d

Threads: #5 pool
0	syscall	/usr/lib/libc.so.6		0x7ffff3d71f89
1	g_cond_wait_until	/usr/lib/libglib-2.0.so.0		0x7ffff2862267
2	??	/usr/lib/libglib-2.0.so.0		0x7ffff27f2779
[... 45 lines cut ...]
5	QThread::exec()	/usr/lib/libQt5Core.so.5		0x7ffff50fe5d4
6	MAVLinkDecoder::run	MAVLinkDecoder.cc	67	0x72af58
7	??	/usr/lib/libQt5Core.so.5		0x7ffff510355e
8	start_thread	/usr/lib/libpthread.so.0		0x7ffff4864374
9	clone	/usr/lib/libc.so.6		0x7ffff3d7627d

Threads: #10 QThread

0	poll	/usr/lib/libc.so.6		0x7ffff3d6d4ed
1	??	/usr/lib/libglib-2.0.so.0		0x7ffff281d9a4
2	g_main_context_iteration	/usr/lib/libglib-2.0.so.0		0x7ffff281dabc
3	QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff534708c
4	QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)	/usr/lib/libQt5Core.so.5		0x7ffff52ec7b2
5	QThread::exec()	/usr/lib/libQt5Core.so.5		0x7ffff50fe5d4
6	??	/usr/lib/libQt5Core.so.5		0x7ffff510355e
7	start_thread	/usr/lib/libpthread.so.0		0x7ffff4864374
8	clone	/usr/lib/libc.so.6		0x7ffff3d7627d

Threads: #11 QXcbEventReader

0	??	/usr/lib/libxcb.so.1		0x7fffedd8e122
1	??	/usr/lib/qt/plugins/platforms/libqxcb.so		0x7fffea2dd3ba
2	??	/usr/lib/libQt5Core.so.5		0x7ffff510355e
3	start_thread	/usr/lib/libpthread.so.0		0x7ffff4864374
4	clone	/usr/lib/libc.so.6		0x7ffff3d7627d

DonLakeFlyer, 2015-04-09:
Onboard parameters widget is a bit of a mess with respect to connecting/disconnecting and multiple uas. It doesn't handle disconnect correctly also switching uas is using list indices which is not a good idea.

DonLakeFlyer, 2015-04-09:
I'll deal with this when I re-write parameter list editor in qml.

mihadyuk, 2015-04-09:
Ok. I'll expect the fix. Thanks for your job.

DonLakeFlyer, 2015-04-15:
Fixed by new parameter editor

closed 2015-04-15 by DonLakeFlyer