mavlink/qgroundcontrol #1436: Parameter loader doesn't always complete load

DonLakeFlyer, 2015-04-09:
When running over Radio you've got a 10-15% chance that the parameter load doesn't finish. It sits at the end waiting for 1 or more params to come down. Not sure why yet. Maybe missing one on the wire and then the re-request logic isn't working correctly.

severinleuenberger, 2015-04-28:
I recognize this behaviour as well, and I download every few days the new qgroundcontrol beta version with your newest fixes. Its still there now. 
After disconnecting and connecting (sometimes 2 times neccessary) it loads correct again. 
But it seems to be a real bug, because I see this at least every 4th restart.

DonLakeFlyer, 2015-04-28:
Are you running latest firmware master? There was a bug in firmware which was causing this.

severinleuenberger, 2015-04-28:
ok, I will just test with another board who has the newest px4 firmware.

DonLakeFlyer, 2015-04-28:
Forget I had entered this issue! Using both latest firmware master and QGC Daily this is fixed.

closed 2015-04-28 by DonLakeFlyer

severinleuenberger, 2015-04-28:
Ok, indeed, I just testet with latest firmware and QGC Daily, and everything is fine again!