mavlink/qgroundcontrol #1454: Look at hosting releases using GitHub release

DonLakeFlyer, 2015-04-14:
To upload:
$ curl -H "Authorization: token TOKEN" \
 -H "Accept: application/vnd.github.manifold-preview" \
 -H "Content-Type: application/zip" \
 --data-binary @build/mac/package.zip \
 "https://uploads.github.com/repos/hubot/singularity/releases/123/assets?name=1.0.0-mac.zip"

dagar, 2015-04-14:
For the stable releases create tags eg "v2.3" and upload the binaries. That should also give me something to base the version (git describe) on for the nightly ppa.

For the development build maybe create a moving tag (delete and retag)?

LorenzMeier, 2015-04-25:
Is this still the plan or are we happy with AWS?

DonLakeFlyer, 2015-04-25:
I do want to look at it at some point. It's not a drop in replacement to the existing TC scripts like AWS was so it's going to take longer.

LorenzMeier, 2015-04-25:
Lets close it until we revisit the overall release process.

closed 2015-04-25 by LorenzMeier