mavlink/qgroundcontrol #1456: Qml Error / module Qt Location is not installed

severinleuenberger, 2015-04-15:
I use the newest developer build 2.4.0 from 14.04.2015. 
And i get everytime the message:

[E] at qrc:///qml/QGroundControl/HUDControls/QGCMapBackground.qml:32 - "qrc:///q
ml/QGroundControl/HUDControls/QGCMapBackground.qml:32:1: module "QtLocation" is
not installed"
[E] at qrc:///qml/QGroundControl/HUDControls/QGCMapBackground.qml:31 - "qrc:///q
ml/QGroundControl/HUDControls/QGCMapBackground.qml:31:1: module "QtPositioning"
is not installed"
[E] at qrc:///qml/QGroundControl/HUDControls/QGCMapBackground.qml:32 - "qrc:///q
ml/QGroundControl/HUDControls/QGCMapBackground.qml:32:1: module "QtLocation" is
not installed"
[E] at qrc:///qml/QGroundControl/HUDControls/QGCMapBackground.qml:31 - "qrc:///q
ml/QGroundControl/HUDControls/QGCMapBackground.qml:31:1: module "QtPositioning"
is not installed"

What could be the cause here?

I am using the QGC on 2 computers, on a Win7 and the other a Win8 64bit. 
Same message on both. 

Best 
Severin

DonLakeFlyer, 2015-04-15:
If you get errors like this. Please post them to the google group instead of creating a github issue.

DonLakeFlyer, 2015-04-15:
We want bugs to start on the email group and then when verified they are real an Issue is created.

DonLakeFlyer, 2015-04-15:
Reopening since I think this is real.
@dogmaphobic I took a quick look at this. I think QGCSetup.pri needs to be updated for the QtLocation modules. Can you look?

dogmaphobic, 2015-04-15:
Sure!

dogmaphobic, 2015-04-15:
Before I go there, when was this changed? If it was a build issue, wouldn't TC fail as well?

DonLakeFlyer, 2015-04-15:
It's an issue for installer versions where Qt is not already on the machine. Since Qt is already on the TC VMs there isn't any real way to test the installer. So even though QGCSetup.pri runs on all builds, if there is something missing there it will only sometimes show up as a problem when you create an installer from it. And then install on a virgin machine.

DonLakeFlyer, 2015-04-15:
Most likely here:
    # Copy Qml libraries
    Q_DIR = $$[QT_INSTALL_QML]
    QML_DIR_WIN = $$replace(Q_DIR, "/", "\\")
    QMAKE_POST_LINK += $$escape_expand(\\n) $$QMAKE_COPY_DIR \"$$QML_DIR_WIN\\QtGraphicalEffects\" \"$$DESTDIR_WIN\\QtGraphicalEffects\"
    QMAKE_POST_LINK += $$escape_expand(\\n) $$QMAKE_COPY_DIR \"$$QML_DIR_WIN\\QtQuick\" \"$$DESTDIR_WIN\\QtQuick\"
    QMAKE_POST_LINK += $$escape_expand(\\n) $$QMAKE_COPY_DIR \"$$QML_DIR_WIN\\QtQuick.2\" \"$$DESTDIR_WIN\\QtQuick.2\"

There must be QtLocation and maybe positioning modules than need to be in here as well?

dogmaphobic, 2015-04-15:
It's on line 168. I will take a better look tomorrow. I'm beat...

dogmaphobic, 2015-04-15:
@severinleuenberger, could you do me a favor? Go to the directory where qgroundcontrol.exe is and run this:
```
dir *.dll > foo.txt
```
And post the contents of foo.txt here so I can take a look.

Thanks!

severinleuenberger, 2015-04-15:
I could not write to foo.txt

C:\Program Files (x86)\qgroundcontrol> dir *.dll > foo.txt
Zugriff verweigert

But i can copy paste it for you here: 

C:\Program Files (x86)\qgroundcontrol> dir *.dll
 Datenträger in Laufwerk C: ist Windows7_OS
 Volumeseriennummer: 3341-0040

 Verzeichnis von C:\Program Files (x86)\qgroundcontrol

03.09.2014  10:16        21'529'088 icudt53.dll
03.09.2014  10:16         1'982'976 icuin53.dll
03.09.2014  10:16         1'355'264 icuuc53.dll
13.04.2015  10:00           151'552 libxbee.dll
05.10.2013  11:38           455'328 msvcp120.dll
05.10.2013  11:38           970'912 msvcr120.dll
12.02.2015  19:02         4'051'456 Qt5Core.dll
05.12.2014  19:38         4'535'296 Qt5Gui.dll
05.12.2014  19:53           321'536 Qt5Location.dll
05.12.2014  19:51           537'088 Qt5Multimedia.dll
05.12.2014  19:52            88'064 Qt5MultimediaWidgets.dll
05.12.2014  19:37           818'688 Qt5Network.dll
05.12.2014  19:41           272'384 Qt5OpenGL.dll
05.12.2014  19:53           157'696 Qt5Positioning.dll
05.12.2014  19:41           264'192 Qt5PrintSupport.dll
05.12.2014  19:45         2'382'848 Qt5Qml.dll
05.12.2014  19:46         2'397'696 Qt5Quick.dll
05.12.2014  19:48            51'200 Qt5QuickWidgets.dll
05.12.2014  19:50           148'480 Qt5Sensors.dll
05.12.2014  19:44            58'368 Qt5SerialPort.dll
05.12.2014  19:37           154'624 Qt5Sql.dll
05.12.2014  19:42           251'392 Qt5Svg.dll
05.12.2014  19:37           133'120 Qt5Test.dll
05.12.2014  20:45        17'378'304 Qt5WebKit.dll
05.12.2014  20:57           197'120 Qt5WebKitWidgets.dll
05.12.2014  19:40         4'397'056 Qt5Widgets.dll
05.12.2014  19:37           150'528 Qt5Xml.dll
13.04.2015  10:00           324'096 SDL.dll
              28 Datei(en),     65'516'352 Bytes
               0 Verzeichnis(se), 50'521'116'672 Bytes frei

If you need the foo.txt file, what can I do to get it work?

Tanks!

dogmaphobic, 2015-04-15:
That's perfect! The proper files are there. I will take a look and see if I can figure out what is going on. 

Thanks!

DonLakeFlyer, 2015-04-15:
@severinleuenberger New installers are posted. Can you try again?

severinleuenberger, 2015-04-15:
installed, looks good again! Thanks!

DonLakeFlyer, 2015-04-15:
@severinleuenberger Awesome. Thanks for reporting and testing.

closed 2015-04-15 by DonLakeFlyer