mavlink/qgroundcontrol #1457: Link management usability issues

LorenzMeier, 2015-04-15:
@dogmaphobic @DonLakeFlyer We are currently doing the equivalent of an user acceptance test to identify low-hanging fruits to improve the user experience (thanks for getting us into a position where that actually starts to make sense!).

Currently users have these two issues:

  * 3DR Radios are not any more auto-discovered, so you need to add them manually
  * On Ubuntu /dev/ttyACM0 and /dev/ttyACM1 always show up in the toolbar menu, even if only one of them is available physically. Not sure if thats due to local setup, Ubuntu or QGC.

DonLakeFlyer, 2015-04-15:
The combo next to connect does not work l like it used to. It is not a list of connected serial ports. It is a list of saved connection which normally are created through preferences. This allows you to create a connection named say "3DR Radio" to a port that is persistent. The item in this list are there even if they are not currently connected.

That said, the board is auto-detected as a Pixhawk and added as a persistent connection automatically. Radios are not auto-detected as a radio since there isn't any easy unique identifying way to do this.

Old code used to just list active serial ports. It had no idea what type the connection was.

Question is whether a new user will be confused since they won't know how it used to work?

dogmaphobic, 2015-04-15:
Yes, you cannot identify a 3DR Radio by simply looking at the manufacturer ID because it's a standard FTDI chip. An old Arduino would show up with a same ID as the 3DR Radio. I guess the only way would be to actively open existing ports and listen for known mavlink traffic in a background thread.

dogmaphobic, 2015-04-17:
I think this is now solved with #1466?

DonLakeFlyer, 2015-04-18:
Yup.

The 3DR radio auto detect is a little odd, but for 90% of users I think it's fine. So good to go. For me it keeps telling me my Telem2 serial port FTDI connection is a radio but that's a developer thing.

closed 2015-04-18 by DonLakeFlyer

dogmaphobic, 2015-04-18:
Yep, that's bound to happen in a few odd cases. My logic was that when you launch QGC, it's by far more likely you have a 3DR Radio than some odd ball device. I guess that makes you the odd ball :)

DonLakeFlyer, 2015-04-18:
Already knew that...